sittner/linuxcnc-ethercat #20: Problem with 64-bits

narogon, 2015-11-30:
Hi Sascha,

Since your last update of config files I get this error in debian wheezy with linuxcnc 2.7.3 64bits. In other system with 32 bits the problem doesn't occur.

make[1]: se ingresa al directorio `/home/fdl/linuxcnc/ethercat/linuxcnc-ethercat/src'
make[2]: se ingresa al directorio `/home/fdl/linuxcnc/ethercat/linuxcnc-ethercat/src'
gcc -o lcec_conf.o -Os -g -I. -I/usr/realtime-3.4-9-rtai-686-pae/include -I. -I/usr/realtime-3.4-9-rtai-686-pae/include -I/usr/include/i386-linux-gnu -U_FORTIFY_SOURCE -D_FORTIFY_SOURCE=0 -fno-math-errno -funsafe-math-optimizations -fno-rounding-math -fno-signaling-nans -fcx-limited-range -mhard-float -DRTAI=3 -fno-fast-math -mieee-fp -fno-unsafe-math-optimizations -DRTAPI -D_GNU_SOURCE -Drealtime -D__MODULE__ -I/usr/include/linuxcnc -URTAPI -U__MODULE__ -DULAPI -Os -c lcec_conf.c
gcc -o lcec_conf lcec_conf.o -Wl,-rpath,/lib -L/lib -llinuxcnchal -lexpat
/usr/bin/ld: /usr/lib/gcc/i486-linux-gnu/4.7/../../../i386-linux-gnu/crti.o: unrecognized relocation (0x2b) in section `.init'
/usr/bin/ld: final link failed: Bad value
collect2: error: ld returned 1 exit status
make[2]: *** [lcec_conf] Error 1
make[2]: se sale del directorio `/home/fdl/linuxcnc/ethercat/linuxcnc-ethercat/src'
make[1]: *** [all] Error 2
make[1]: se sale del directorio `/home/fdl/linuxcnc/ethercat/linuxcnc-ethercat/src'
make: *** [all] Error 2

sirop, 2015-11-30:
My 2 cents:
No problems on Xenomai kernel.

sirop, 2015-12-15:
If you are on 64 bit , you should have intalled:
````
sudo apt-get install linux-image-rtai.x86-amd64
````
`realtime-3.4-9-rtai-686-pae` is 32 bit version.  

Do not know if this is the only reason of your issue, but it is at least strange.

narogon, 2015-12-15:
Sirop, first of all thanks for your replies. 

I don't think I made myself clear. The SO is the same in both PCs. Linux Debian 3.4-9-rtai-686-pae  specifically linuxcnc 2.7 branch. The one in which i haven't any problem is an old laptop but the other is a new PC 64 bits capable. 

In any case, I have made lot of tests and I have no idea where the problem is. I have tested 5 computers with the same linux version and no problems in any of them. The only difference is in the hardware so I suppose it could be there. 

For now I'm compiling the components in other computer and I install them in the problematic PC without #make clean so it only copies the files.

Thanks for your time
Best regards

closed 2016-03-23 by narogon