sittner/linuxcnc-ethercat #31: EL7041-1000 vs EL7041

koppi, 2016-07-07:
In https://forum.linuxcnc.org/forum/24-hal-components/22346-ethercat-hal-driver?start=310#76884 a question from DaPeace arised whether the EL7041 is supported by linuxcnc-ethercat.

When I wrote https://github.com/sittner/linuxcnc-ethercat/blob/master/src/lcec_el7041_1000.c I only had the EL7041-1000 module at hand, but the code in lcec_el7041_1000.c should probably also work with the EL7041 module.

sittner, 2018-03-09:
Could somebody verify this?

scottlaird, 2019-12-07:
I have an EP7041, it seems to probe okay for me with EL7041-1000 in ethercat.xml, but I haven't hooked a motor or encoder up yet.  I have a couple different EP7041 versions, and I'd be kind of surprised if they don't work.

I'm working on adding support for a number of additional modules (see https://github.com/scottlaird/linuxcnc-ethercat), I'll send a pull request cleaning up the EL/EP7041 module naming along with the other changes once things are ready.

closed 2025-02-16 by koppi