sittner/linuxcnc-ethercat #38: mffu syncmanger delta asda-a2-e

fupe, 2017-09-26:
HI,
I try to run delta driver asda-a2-e but without result.
I can see all device (ethercat slaves) i can read write thrue CoE, but when I run linuxcnc, dmesg shows 
problem with syncmanager for SM2 and SM3 because cannot read info for MFFU adress, size etc.
is this info in E2PROM? 
Is it possible to write this info into xml config file for lcec_conf?
Thanks. Martin

sittner, 2018-03-09:
Do you have some logs? If I remember correctly you have to configure the pdo mapping with the delta tool to met the expected one:

static ec_pdo_entry_info_t lcec_deasda_in[] = {
   {0x6041, 0x00, 16}, // Status Word
   {0x606C, 0x00, 32}, // Current Velocity
   {0x6064, 0x00, 32}, // Current Position
   {0x2511, 0x00, 32}  // external encoder
};

static ec_pdo_entry_info_t lcec_deasda_out[] = {
   {0x6040, 0x00, 16}, // Control Word
   {0x60FF, 0x00, 32}  // Target Velocity
};

static ec_pdo_info_t lcec_deasda_pdos_out[] = {
    {0x1602,  2, lcec_deasda_out}
};

static ec_pdo_info_t lcec_deasda_pdos_in[] = {
    {0x1a02, 4, lcec_deasda_in}
};

static ec_sync_info_t lcec_deasda_syncs[] = {
    {0, EC_DIR_OUTPUT, 0, NULL},
    {1, EC_DIR_INPUT,  0, NULL},
    {2, EC_DIR_OUTPUT, 1, lcec_deasda_pdos_out},
    {3, EC_DIR_INPUT,  1, lcec_deasda_pdos_in},
    {0xff}
};

fupe, 2018-03-12:
Hi Sascha,
thanks for reply. I have no logs, it is old thread. I managed it.
There was some strengh in delsta ASDA-A2 driver. EEPROM has only 256B and there was only this info
 
CoE ....................... True
EoE ....................... False
FoE ....................... False
SoE ....................... False
VoE ....................... False
No info about name, nothing.
I used siitool for look inside EEPROM.
I cannot changed PDO mapings from linuxcnc. (dmesg wrote some error info about it)
I wrote new bin file for EEPROM with 
Enable PDO Assign: ........... yes
Enable PDO Configuration: .... yes
But mainly i have to wrote info about SM and FMMU in to EEPROM.
min size, max size, start address control byte etc. This information are in xml file from DELTA
<Sm MinSize="34" MaxSize="192" DefaultSize="128" StartAddress="#x1000" ControlByte="#x36" Enable="1">MBoxOut</Sm>
<Sm MinSize="34" MaxSize="192" DefaultSize="128" StartAddress="#x10C0" ControlByte="#x32" Enable="1">MBoxIn</Sm>
<Sm MinSize="1" MaxSize="256" DefaultSize="6" StartAddress="#x1180" ControlByte="#x24" Enable="1">Outputs</Sm>
<Sm MinSize="1" MaxSize="256" DefaultSize="6" StartAddress="#x1480" ControlByte="#x00" Enable="1">Inputs</Sm>

 FMMU 
<Fmmu>Outputs</Fmmu>
<Fmmu>Inputs</Fmmu>
<Fmmu>MBoxState</Fmmu>
Now linuxcnc is able to create DOMAIN and work fine with drivers.
Thanks.
My first question was.... Where ethercat software can find out information about SM and FMMU (min size, max size, start address control byte) when they are not in EEPROM.
Martin
Sorry for my english

closed 2018-03-12 by sittner

narogon, 2019-01-28:
I've suffering the same problem using delta asda-a2, now from fw version 1.647 they have blocked the eeprom updating.  
Could you add this patch to ec-debianize? 
https://sourceforge.net/u/uecasm/etherlab-patches/ci/default/tree/features/sii-file/

I think this message in etherlab maillist is the origin of the patch:
http://lists.etherlab.org/pipermail/etherlab-users/2011/001464.html