sittner/linuxcnc-ethercat #42: hello, scale AX5000

chimeno86, 2018-03-01:
Hello, can you help me with the enlargement of the scale for the AX5000 in speed mode ?, tried to follow the example of the file Stmds5K.c but I can not get it to work well, can you help me?
Thank you.

sittner, 2018-03-02:
Just wait for a few days. I currently integrate an ax5206. Perhaps the questions will dissapear :-)

Regards
Sascha

chimeno86, 2018-03-02:
Hello Sascha,
I thank you, thank you very much for the help.

regards
Chimeno

sittner, 2018-03-22:
Hi,

the AX5200 stuff is commited, please check if that helps.

Regards
Sascha

chimeno86, 2018-03-26:
Hello Sasha,
thank you I will make a clean installation with your new code and I will tell you.

regards
Chimeno

chimeno86, 2018-03-27:
hi Sasha,
I'm with your code, I have a couple of questions the same is error:
signal, post-fb should be float type, linuxcnc does not allow S32, connected axis.0.motor-pos-fb
signal, vel-cmd should be float type, linuxcnc does not allow S32 connected x-pid.output

regards
Chimeno

sittner, 2018-03-28:
They _are_ float:

https://github.com/sittner/linuxcnc-ethercat/blob/45f0d55fc3be4d24cf0ec4355f342088713304c3/src/lcec_class_enc.c#L32

https://github.com/sittner/linuxcnc-ethercat/blob/45f0d55fc3be4d24cf0ec4355f342088713304c3/src/lcec_ax5200.c#L70

chimeno86, 2018-03-29:
Hello Sasha,
I was silly, use the example linuxcnc-ethercat / examples / initcmds / ethercat-conf.xml, now I already use the specific module lcec_ax5000, I'll try to see how it works ...
regards
Chimeno

chimeno86, 2018-03-31:
hi Sasha,
I am ultilizando and I have a problem that already had in the old version of linuxcnc-ethercat, the module AX5206 is connected and disconnected during the time, you know what I can do?
The configuration that I have is:
  <slave idx = "0" type = "AX5206">
 
       <dcConf assignActivate = "730" sync0Cycle = "250000" sync0Shift = "50000" sync1Cycle = "750000" sync1Shift = "250000" />
     </ slave>

regards
Chimeno

chimeno86, 2018-07-20:
hello currently the problem of synchronism placing a buffer in the reading of the signals, is the only way found to not lose the connection with the AX5000, example:
pos_fb = EC_READ_S32 (& pd [chan-> pos_fb_pdo_os]);
if (pos_fb! = 0) {
* (chan-> pos_fb) = pos_fb;
}

Thank you

chimeno86, 2018-07-20:
hi Sasha,
I would like to know if it is possible that you can add "lcec_write_idn"? I would like to be able to unlock the motor brake using the functions P-0-0096, what was tested in the initial configuration and it works well, tried to use the ecrt_master_write_idn function, but it does not work for me, do not know if you can help me? 
Thank you
Chimeno

robochime, 2019-06-07:
solved thanks to the patch.

closed 2019-06-07 by robochime