sittner/linuxcnc-ethercat #49: Segmentation fault during shutdown

sirop, 2018-03-10:
Hello.

The essence of this discussion https://github.com/koppi/mk/issues/4 is that the build command
for the user space case is incomplete as we see it in https://github.com/sittner/linuxcnc-ethercat/blob/master/src/user.mk .

Namely:
>  missing essential steps: objcopy - extraction of symbols in the .rtapi_export section, creation of a linker script fragment, final link

Do you have any simple remedy for this?

Tested with machinekit.

sittner, 2018-03-10:
I guess the discussed problem affected the RT part, not the user space part.
I've tried to fix this by using Makefile.modinc to compile the RT component, which is definitely the nicer method :-):

https://github.com/sittner/linuxcnc-ethercat/tree/fix-build-script

Please let me know, if that woks for you.
Do you think this will affect the segfault?

sittner, 2018-03-10:
This was too fast. I have to find a way to link the additional needed libraries.

sittner, 2018-03-10:
OK, LDFLAGS did the job. Sorry for the interrupt...

sirop, 2018-03-10:
@sittner 
Ok, give me a day or two to check this.