sittner/linuxcnc-ethercat #55: could not insert module lcec.ko

ilyaq, 2018-03-30:
Hello sittner
Thank you for your work
I installed your driver using deb http://repo.modusoft.de/linuxcnc/wheezy ./
My problem is running the program using `loadrt lcec` in the hal file. 
I getting that:
Error: could not insert module /usr/realtime-3.4-9-rtai-686-pae/modules/linuxcnc/lcec.ko: Unknown symbol in module
[linuxcnc_debug.txt](https://github.com/sittner/linuxcnc-ethercat/files/1863748/linuxcnc_debug.txt)
[linuxcnc_print.txt](https://github.com/sittner/linuxcnc-ethercat/files/1863749/linuxcnc_print.txt)

I know little about this and I hope for Your help. Would you mind pointing me in the right direction?

chimeno86, 2018-03-31:
Hi, ilyaq
Are you using RTAI, are not you? I think the new version only works for RP-PREEMPT ..
regards
Chimeno

chimeno86, 2018-04-02:
hi ilyaq
important is "debian 3.4-9-rtai-686-pae" your core kernel is rtai, there are two options or re-copy the kenner or if you do not have much experience install linuxcnc again from this location that is formed by 4.9.5 with core RT-PREEMPT http://www.linuxcnc.org/testing-stretch-rtpreempt/
regards
Chimeno

ilyaq, 2018-04-02:
Hi, chimeno
I used the second option and installed linuxcnc which you advised me. It helped, thank you! 
But now I get another error. I am trying to run swm-fm45a example 

`To run under a debugger with I/O, use e.g.,`
`    sudo env RTAPI_UID=`id -u` RTAPI_FIFO_PATH=$HOME/.rtapi_fifo gdb rtapi_app`
`./swm-fm45a.hal:16: waitpid failed /usr/bin/rtapi_app trivkins`
`./swm-fm45a.hal:16: /usr/bin/rtapi_app exited without becoming ready`
`./swm-fm45a.hal:16: insmod for trivkins failed, returned -1`

I'm not sure I installed the HAL-driver correctly

regards
ilyaq

ilyaq, 2018-04-04:
Hi, chinemo
First, Thank you for your help
May I clarify? What configuration do you mean? I can't use `loadrt lcec` in my hal file. I get the error
Do you know what this is about? XML, ini, hal or are there other configurations I could skip?
thanks 
ilya

chimeno86, 2018-04-04:
hi ilyaq,
what modules do you have?

.hal file
# load ethercat config parser
loadusr -W lcec_conf ethercat-conf.xml "edit with the modules you have"
# load ethercat realtime module
loadrt lcec
addf lcec.read-all servo-thread
addf lcec.write-all servo-thread

regards
Chimeno

ilyaq, 2018-04-05:
Hi chimeno,
i have two servo drive Delta ASDA A2-E.
If I don't edit .xml file, then the program will not start?
I mean. I used  `loadusr -W lcec_conf ethercat-conf.xml`  and `loadrt lcec` in .hal file, but I get an error when the program reaches the line containing `loadrt lcec`. But I didn't edit the .xml file. Can whether the reason be in this?
Thanks
Ilya

ilyaq, 2018-04-10:
Hi chimeno,
You've already helped me a lot. And I'm ashamed of myself, but I still need help.
I've been trying to figure it out on my own, but so far, it's not working.

I get this error message:
./swm-fm45a-io.hal:41: parameter or pin 'lcec.0.3A1.srv-pos-scale' not found
Do you know what this is about?

Regards,
Ilya

chimeno86, 2018-04-11:
Hello Ilya,
   if it is a common error, it means that in your .hal file you have connected a pin to 'lcec.0.3A1.srv-pos-scale that does not exist, looking at the file src / lcec_deasda.c I think it would be "lcec.0.3A1. pos-scale ", prove if it works for you now.

regards
Chimeno

ilyaq, 2018-04-12:
Hello Chimeno,
Yes, It worked! Many thanks!
Following your example, i found in file lcec_deasda.c this parameter - pos-scale  and used it instead of srv-pos-scale.

Now I get the following error:
./swm-fm45a-io.hal:70: Pin 'lcec.0.3A1.srv-pos-extenc' does not exist
In the hal file is written as follows:
`# motor encoder position is used as joint`
`# position feedback`
`net x-pos-fb <= lcec.0.3A1.srv-pos-extenc`

I tried to find information in the lcec_deasda.c and lcec_deasda.h files, and tried using different parameters from them, but so far without success.

regards
Ilya

chimeno86, 2018-04-12:
Hello Ilya,
It's the same mistake that happened before, I think it would be something like this:

# motor encoder position is used as joint
# position feedback
net x-post-fb <= lcec.0.3A1.srv-vel-fb

regards
Chimeno

chimeno86, 2018-04-16:
Hello Ilya,
you can look at the issue of the "generic" driver, it can help you with the tests.
regards
Chimeno

ilyaq, 2018-04-18:
Hello chimeno
I found the right pin. 
If anyone needs it, then you need to use the following
`net x-post-fb <= lcec.0.3A1.enc-pos`

I am very grateful to you, chimeno!
regards
Ilya

chimeno86, 2018-04-18:
Hello Ilya,
I'm glad you finally work well! It has been a pleasure to help you!
regards
Chimeno