sittner/linuxcnc-ethercat #57: ./debian/linuxcnc-ethercat/usr/lib/linuxcnc/modules/ directory is not created while running dpkg-buildpackage

tstaroveski, 2018-05-17:
Dear Mr. Ittner,

Thank you very much for the great work!

I would like to report a problem which occurs while building the packagage using dpkg-buildpackage command. The problem seems to be related to one of the Makefiles (presumably src/realtime.mk). 

After cloning repository and running dpkg-buildpackage, I get the following error:
....
make[3]: Entering directory '/home/sarma/src/linuxcnc-ethercat/linuxcnc-ethercat/src'
mkdir -p /home/sarma/src/linuxcnc-ethercat/linuxcnc-ethercat/debian/linuxcnc-ethercat/usr/bin
cp lcec_conf /home/sarma/src/linuxcnc-ethercat/linuxcnc-ethercat/debian/linuxcnc-ethercat/usr/bin/
make[3]: Leaving directory '/home/sarma/src/linuxcnc-ethercat/linuxcnc-ethercat/src'
make[3]: Entering directory '/home/sarma/src/linuxcnc-ethercat/linuxcnc-ethercat/src'
cp lcec.so /home/sarma/src/linuxcnc-ethercat/linuxcnc-ethercat/debian/linuxcnc-ethercat/usr/lib/linuxcnc/modules/
cp: cannot create regular file '/home/sarma/src/linuxcnc-ethercat/linuxcnc-ethercat/debian/linuxcnc-ethercat/usr/lib/linuxcnc/modules/': No such file or directory
/usr/share/linuxcnc/Makefile.modinc:104: recipe for target 'install' failed
make[3]: *** [install] Error 1
make[3]: Leaving directory '/home/sarma/src/linuxcnc-ethercat/linuxcnc-ethercat/src'
Makefile:8: recipe for target 'install' failed
make[2]: *** [install] Error 2
make[2]: Leaving directory '/home/sarma/src/linuxcnc-ethercat/linuxcnc-ethercat/src'
Makefile:11: recipe for target 'install' failed
make[1]: *** [install] Error 2
make[1]: Leaving directory '/home/sarma/src/linuxcnc-ethercat/linuxcnc-ethercat'
dh_auto_install: make -j1 install DESTDIR=/home/sarma/src/linuxcnc-ethercat/linuxcnc-ethercat/debian/linuxcnc-ethercat AM_UPDATE_INFO_DIR=no returned exit code 2
debian/rules:13: recipe for target 'binary' failed
make: *** [binary] Error 2
dpkg-buildpackage: error: fakeroot debian/rules binary gave error exit status 2

It would seem that linuxcnc-ethercat/usr/lib/linuxcnc/modules/ directory is not created during package build.

tstaroveski, 2018-05-17:
Many thanks for the quick response.

Right now I am using 4.9.98-rt76 (SMP PREEMPT), on Linux MINT 18.3 x64, but I think I observed similar effects on Debian Stretch with linux-image-4.9.0-6-rt-amd64 kernel package. I will try to repeat the build process on Debian Stretch again and post back.

ec-debianize and linuxcnc both compiled sucsesfully and work well on this kernel.  

*** To make sure that ethercat drivers are loaded, I ran the following commands (single slave connected to the master):
```
~/src/ec-debianize/ec-debianize $ lsmod |grep ec_
ec_generic             16384  0
ec_master             266240  1 ec_generic

~/src/ec-debianize/ec-debianize $ ethercat slaves
0  0:0  PREOP  +  DP/DZ Series; 1-Axis

~/src/ec-debianize/ec-debianize $ sudo dmesg |grep EtherCAT
[    4.907979] EtherCAT: Master driver 1.5.2 unknown
[    4.908054] EtherCAT: 1 master waiting for devices.
[    4.909103] ec_generic: EtherCAT master generic Ethernet device module 1.5.2 unknown
[    4.909115] EtherCAT: Accepting XX:XX:XX:XX:XX:XX as main device for master 0.
[    4.988904] EtherCAT 0: Starting EtherCAT-IDLE thread.
[   10.184061] EtherCAT 0: Link state of ecm0 changed to UP.
[   10.192110] EtherCAT WARNING 0: 1 datagram TIMED OUT!
[   10.252470] EtherCAT 0: 1 slave(s) responding on main device.
[   10.252473] EtherCAT 0: Slave states on main device: PREOP.
[   10.252759] EtherCAT 0: Scanning bus.
[   10.468038] EtherCAT 0: Bus scanning completed in 216 ms.
[   10.468040] EtherCAT 0: Using slave 0 as DC reference clock.
```

tstaroveski, 2018-05-18:
I can confirm the same error when building on Debian Stretch with stock rt-preempt kernel (4.9.0-6-rt-amd64 #1 SMP PREEMPT RT Debian 4.9.88-1+deb9u1 (2018-05-07))
```
...
cp lcec_conf /home/user/src/linuxcnc-ethercat/linuxcnc-ethercat/debian/linuxcnc-ethercat/usr/bin/
make[3]: Leaving directory '/home/user/src/linuxcnc-ethercat/linuxcnc-ethercat/src'
make[3]: Entering directory '/home/user/src/linuxcnc-ethercat/linuxcnc-ethercat/src'
cp lcec.so /home/user/src/linuxcnc-ethercat/linuxcnc-ethercat/debian/linuxcnc-ethercat/usr/lib/linuxcnc/modules/
cp: cannot create regular file '/home/user/src/linuxcnc-ethercat/linuxcnc-ethercat/debian/linuxcnc-ethercat/usr/lib/linuxcnc/modules/': No such file or directory
/usr/share/linuxcnc/Makefile.modinc:104: recipe for target 'install' failed
make[3]: *** [install] Error 1
make[3]: Leaving directory '/home/user/src/linuxcnc-ethercat/linuxcnc-ethercat/src'
Makefile:8: recipe for target 'install' failed
make[2]: *** [install] Error 2
make[2]: Leaving directory '/home/user/src/linuxcnc-ethercat/linuxcnc-ethercat/src'
Makefile:11: recipe for target 'install' failed
make[1]: *** [install] Error 2
make[1]: Leaving directory '/home/user/src/linuxcnc-ethercat/linuxcnc-ethercat'
dh_auto_install: make -j1 install DESTDIR=/home/user/src/linuxcnc-ethercat/linuxcnc-ethercat/debian/linuxcnc-ethercat AM_UPDATE_INFO_DIR=no returned exit code 2
debian/rules:13: recipe for target 'binary' failed
make: *** [binary] Error 2
dpkg-buildpackage: error: fakeroot debian/rules binary gave error exit status 2
```

tstaroveski, 2018-05-29:
Here is another update:

I get the same error as before when trying to compile the code with latest commit (407698b )

make && make install (without using dpkg-buildpackage) seems to compile without errors (same kernels used as before):

```
~/src/linuxcnc-ethercat/linuxcnc-ethercat $ make
make[1]: Entering directory '/home/user/src/linuxcnc-ethercat/linuxcnc-ethercat/src'
make[2]: Entering directory '/home/user/src/linuxcnc-ethercat/linuxcnc-ethercat/src'
gcc -o lcec_conf.o -Os -g -I. -I/home/user/src/linuxcnc/linuxcnc-dev/src/include -DUSPACE -fno-fast-math -mieee-fp -fno-unsafe-math-optimizations -DRTAPI -D_GNU_SOURCE -Drealtime -D__MODULE__ -I/usr/include/linuxcnc -DSIM -fPIC -URTAPI -U__MODULE__ -DULAPI -Os -c lcec_conf.c
gcc -o lcec_conf_util.o -Os -g -I. -I/home/user/src/linuxcnc/linuxcnc-dev/src/include -DUSPACE -fno-fast-math -mieee-fp -fno-unsafe-math-optimizations -DRTAPI -D_GNU_SOURCE -Drealtime -D__MODULE__ -I/usr/include/linuxcnc -DSIM -fPIC -URTAPI -U__MODULE__ -DULAPI -Os -c lcec_conf_util.c
gcc -o lcec_conf_icmds.o -Os -g -I. -I/home/user/src/linuxcnc/linuxcnc-dev/src/include -DUSPACE -fno-fast-math -mieee-fp -fno-unsafe-math-optimizations -DRTAPI -D_GNU_SOURCE -Drealtime -D__MODULE__ -I/usr/include/linuxcnc -DSIM -fPIC -URTAPI -U__MODULE__ -DULAPI -Os -c lcec_conf_icmds.c
gcc -o lcec_conf lcec_conf.o lcec_conf_util.o lcec_conf_icmds.o  -Wl,-rpath,/lib -L/lib -llinuxcnchal -lexpat
make[2]: Leaving directory '/home/user/src/linuxcnc-ethercat/linuxcnc-ethercat/src'
make[2]: Entering directory '/home/user/src/linuxcnc-ethercat/linuxcnc-ethercat/src'
Compiling realtime lcec_main.c
Compiling realtime lcec_class_enc.c
Compiling realtime lcec_generic.c
Compiling realtime lcec_ax5200.c
Compiling realtime lcec_el1xxx.c
Compiling realtime lcec_el1252.c
Compiling realtime lcec_el2521.c
Compiling realtime lcec_el2xxx.c
Compiling realtime lcec_el2202.c
Compiling realtime lcec_el31x2.c
Compiling realtime lcec_el3255.c
Compiling realtime lcec_el40x1.c
Compiling realtime lcec_el40x2.c
Compiling realtime lcec_el41x2.c
Compiling realtime lcec_el41x4.c
Compiling realtime lcec_el5101.c
Compiling realtime lcec_el5151.c
Compiling realtime lcec_el5152.c
Compiling realtime lcec_el7041_1000.c
Compiling realtime lcec_el7211.c
Compiling realtime lcec_el7342.c
Compiling realtime lcec_el95xx.c
Compiling realtime lcec_em7004.c
Compiling realtime lcec_stmds5k.c
Compiling realtime lcec_deasda.c
Linking lcec.so
make[2]: Leaving directory '/home/user/src/linuxcnc-ethercat/linuxcnc-ethercat/src'
make[1]: Leaving directory '/home/user/src/linuxcnc-ethercat/linuxcnc-ethercat/src'

~/src/linuxcnc-ethercat/linuxcnc-ethercat $ sudo make install
make[1]: Entering directory '/home/user/src/linuxcnc-ethercat/linuxcnc-ethercat/src'
make[2]: Entering directory '/home/user/src/linuxcnc-ethercat/linuxcnc-ethercat/src'
mkdir -p /usr/bin
cp lcec_conf /usr/bin/
make[2]: Leaving directory '/home/user/src/linuxcnc-ethercat/linuxcnc-ethercat/src'
make[2]: Entering directory '/home/user/src/linuxcnc-ethercat/linuxcnc-ethercat/src'
cp lcec.so /usr/lib/linuxcnc/modules/
make[2]: Leaving directory '/home/user/src/linuxcnc-ethercat/linuxcnc-ethercat/src'
make[1]: Leaving directory '/home/user/src/linuxcnc-ethercat/linuxcnc-ethercat/src'
make[1]: Entering directory '/home/user/src/linuxcnc-ethercat/linuxcnc-ethercat/examples'
mkdir -p /usr/share/linuxcnc-ethercat/examples
cp -R generic-complex swm-fm45a /usr/share/linuxcnc-ethercat/examples
make[1]: Leaving directory '/home/user/src/linuxcnc-ethercat/linuxcnc-ethercat/examples'
```
However, when trying to start the hal script (which works quite fine on older rtai build), I get the following error:
```
$halrun -I srm_amc_fullpdo.hal 
Note: Using POSIX realtime
lcec: dlopen: /usr/lib/linuxcnc/modules/lcec.so: undefined symbol: ecrt_slave_config_sdo
srm_amc_fullpdo.hal:7: waitpid failed /usr/bin/rtapi_app lcec
srm_amc_fullpdo.hal:7: /usr/bin/rtapi_app exited without becoming ready
srm_amc_fullpdo.hal:7: insmod for lcec failed, returned -1
halcmd: exit
```
I am trying to get the linuxcnc-ethercat package working on these kernels since support for rtai builds was dropped in latest versions of ec-debianize. Older rt-preempt kernels do not seem to make much sense since older rtai kernels perform much better.  

Kind regards

gausstop, 2019-04-01:
src/user.mk

install: lcec_conf
	mkdir -p $(DESTDIR)$(EMC2_HOME)/bin
	mkdir -p $(DESTDIR)$(RTLIBDIR)/
	cp lcec_conf $(DESTDIR)$(EMC2_HOME)/bin/