sittner/linuxcnc-ethercat #61: libethercat not linked

gbcwbz, 2018-10-09:
* use master branch, run `loadrt lcec` in halcmd environment, an error occur that -- ` Undefined symbol ecrt_slave_config_sdo`
* I found that lcec.so is not linked with libethercat.so by running `ldd lcec.so`
* I reverted commit 7b1e4bb0bb0b5886486b98fc2511b2333402ad0d , and it works
* my kernel is 4.4.12-rt20

sirop, 2018-11-17:
```
pi@raspberrypi:~/machinekit-hal/rtlib/rt-preempt $ ldd  lcec.so
        linux-vdso.so.1 (0x7efcf000)
        /usr/lib/arm-linux-gnueabihf/libarmmem.so (0x76f2e000)
        liblinuxcnchal.so.0 => /home/pi/machinekit-hal/lib/liblinuxcnchal.so.0 (0x76eee000)
        libethercat.so.1 => /usr/lib/arm-linux-gnueabihf/libethercat.so.1 (0x76ebe000)
        libgcc_s.so.1 => /lib/arm-linux-gnueabihf/libgcc_s.so.1 (0x76e91000)
        libc.so.6 => /lib/arm-linux-gnueabihf/libc.so.6 (0x76d52000)
        libmachinetalk-pb2++.so.0 => /home/pi/machinekit-hal/lib/libmachinetalk-pb2++.so.0 (0x76c0e000)
        libstdc++.so.6 => /usr/lib/arm-linux-gnueabihf/libstdc++.so.6 (0x76ac6000)
        libdl.so.2 => /lib/arm-linux-gnueabihf/libdl.so.2 (0x76ab3000)
        librt.so.1 => /lib/arm-linux-gnueabihf/librt.so.1 (0x76a9c000)
        libprotobuf.so.10 => /usr/lib/arm-linux-gnueabihf/libprotobuf.so.10 (0x7687d000)
        libpthread.so.0 => /lib/arm-linux-gnueabihf/libpthread.so.0 (0x76854000)
        libczmq.so.4 => /usr/lib/arm-linux-gnueabihf/libczmq.so.4 (0x767e6000)
        libzmq.so.5 => /usr/lib/arm-linux-gnueabihf/libzmq.so.5 (0x7676c000)
        /lib/ld-linux-armhf.so.3 (0x76f79000)
        libm.so.6 => /lib/arm-linux-gnueabihf/libm.so.6 (0x766ed000)
        libz.so.1 => /lib/arm-linux-gnueabihf/libz.so.1 (0x766c6000)
        libuuid.so.1 => /lib/arm-linux-gnueabihf/libuuid.so.1 (0x766b2000)
        libsystemd.so.0 => /lib/arm-linux-gnueabihf/libsystemd.so.0 (0x76638000)
        libsodium.so.18 => /usr/lib/arm-linux-gnueabihf/libsodium.so.18 (0x765ce000)
        libpgm-5.2.so.0 => /usr/lib/arm-linux-gnueabihf/libpgm-5.2.so.0 (0x7656e000)
        libselinux.so.1 => /lib/arm-linux-gnueabihf/libselinux.so.1 (0x7653b000)
        liblzma.so.5 => /lib/arm-linux-gnueabihf/liblzma.so.5 (0x7650a000)
        liblz4.so.1 => /usr/lib/arm-linux-gnueabihf/liblz4.so.1 (0x764e9000)
        libgcrypt.so.20 => /lib/arm-linux-gnueabihf/libgcrypt.so.20 (0x76418000)
        libpcre.so.3 => /lib/arm-linux-gnueabihf/libpcre.so.3 (0x7639f000)
        libgpg-error.so.0 => /lib/arm-linux-gnueabihf/libgpg-error.so.0 (0x7637f000)
```

```
pi@raspberrypi:~/linuxcnc-ethercat $ git log | head -10
commit b02252a79d11f1c9dd5884bac1a53743b3818d39
Merge: 22bcaa9 333dec5
Author: Sascha Ittner <sascha.ittner@modusoft.de>
Date:   Fri Aug 17 16:37:32 2018 +0200

    Merge pull request #60 from sittner/add-fsoe-support

    Add fsoe support

commit 22bcaa9a5c0f41beffa41a6b56e7be30a5b878f3
pi@raspberrypi:~/linuxcnc-ethercat $ uname -a
Linux raspberrypi 4.14.74-rt44-v7-custom #1 SMP PREEMPT RT Sun Nov 11 19:17:06 CET 2018 armv7l GNU/Linux
pi@raspberrypi:~/linuxcnc-ethercat $
```

So far no problems:   `libethercat.so.1 => /usr/lib/arm-linux-gnueabihf/libethercat.so.1 (0x76ebe000)` is included...

sirop, 2018-11-17:
@gbcwbz

My guess is that the installation of linuxcnc-ethercat relies on https://github.com/sittner/ec-debianize .
Did you use the debianized installation?

gbcwbz, 2018-11-20:
@sirop No, I installed EtherCAT master manually， maybe that's the reason. Thanks.

shameless, 2018-12-28:
@gbcwbz hey Hi I just have the same problem like you have listed up there, and when I ldd lcec.so, I did not find that it is linked with ethercat. I install etherlab manually.
Can you give me some advices, thank you, regards

gbcwbz, 2018-12-28:
@shameless I reverted commit 7b1e4bb
This commit changed make file, and used linuxcnc's Makefile.modinc, which maybe not correct in your system.

pamxy, 2018-12-28:
I have this problem too! I use "Linux ubuntu 3.14.79-rt85 #1 SMP PREEMPT RT" , etherlab master and this master. And I found the Tag:v0.9.1 's src/realtime.mk can solve it!

pamxy, 2018-12-28:
> @shameless I reverted commit [7b1e4bb](https://github.com/sittner/linuxcnc-ethercat/commit/7b1e4bb0bb0b5886486b98fc2511b2333402ad0d)
> This commit changed make file, and used linuxcnc's Makefile.modinc, which maybe not correct in your system.

So I found that my problem was caused by you changing the realtime.mk file！  :-P

shameless, 2018-12-28:
@gbcwbz  Actually, there's a problem that I did not figure it out. The problem is that when I enter into the directory of the commit you reverted, make it, and there's a message: ***halcompile/comp not executable or not found. So I add the path "linuxcnc/bin" to /etc/profile, and then make it, it is okay for this time. But I can not start linuxcnc, even I do the ./scripts/rip-environmant first(message: this scripts only needs to be run once per shell session), linuxcnc is still not working. 
Regards.

gbcwbz, 2018-12-28:
@shameless You only need to revert that single commit. How did you install linuxcnc ? I installed by a debian package in linuxcnc forum [Ubuntu PPA Repository](https://forum.linuxcnc.org/9-installing-linuxcnc/34998-ubuntu-ppa-repository#115262)

shameless, 2018-12-29:
@gbcwbz There's nothing wrong in the building process, but when I check ldd lcec.so, the message:
linux-vdso.so.1 => (0x00007ffed138e000)
libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007fc43f3ac000)
/lib64/ld-linux-x86-64.so.2 (0x00007fc43f989000)

That's all, nothing else.

sirop, 2019-03-19:
Built linuxcnc-ethercat and etherlab master these days. Both manually (not debianized) on Debian Stretch, Machinekit.
Setting `export LDFLAGS=-L/usr/local/etherlab/lib`  helped with linking `libethercat.so`  into `lcec.so`.

gbcwbz, 2019-09-27:
- The reason is  the order of libraries in the link line.
  - In [Makefile.modinc](https://github.com/machinekit/machinekit-hal/blob/09405ced2a782751107012fab3eb27291264e99c/src/Makefile.modinc.in#L115), 
```c
$(Q)$(CC) -shared -Bsymbolic $(LDFLAGS) -Wl,--version-script,$*.ver -o $@ $^ $(EXTRA_LDFLAGS)
```
`LDFLAGS` includes `-llinuxcnchal -    lethercat`,  if we put `LDFLAGS` at the end of line, it would link successfully
  - I used `Makefile.modinc` from machinekit-hal,  the file from linuxcnc is the same [Makefile.modinc.in](https://github.com/LinuxCNC/linuxcnc/blob/0f91c553a238c3f5e8a52285044761c2dcfd7de5/src/Makefile.modinc.in#L122)

wujianya, 2023-11-01:
My current system is using xenomai real-time kernel, and the same problem also occurs, has anyone solved it?