sittner/linuxcnc-ethercat #65: I wonder how to use linuxcnc-ethercat?Could you please give me a direction for use？

liuzq71, 2018-12-05:


sittner, 2019-03-18:
You are highly welcome to write a good documentation on this issue. I have no time for this task unfortunately. So the code and the forums has to suffice for that at the moment (I'm not very glad with that situation).

liuzq71, 2019-05-04:
how to write the hal and xml files to drive a servo in linuxcnc?would you pls modify them to fit it for me?

myself:
hal:
-------------------------------------------------------------------------
# Erstellt von stepconf am Tue Jul 16 23:06:39 2013
# Änderungen an dieser Datei werden beim nächsten
# Aufruf von stepconf überschrieben.

#loadusr -W lcec_conf /home/pi/linuxcnc/configs/sim.axis/deltaPos.xml
loadusr -W lcec_conf ./is620nPos.xml
loadrt lcec

addf lcec.read-all servo-thread
addf motion-command-handler servo-thread
addf motion-controller servo-thread
addf lcec.write-all servo-thread

#set the control word to 6 and the state will be ready to switch on.
setp lcec.0.0.drivecontrol-1 1
setp lcec.0.0.drivecontrol-2 1

#net Xachse_fb lcec.0.0.pos => joint.0.motor-pos-fb
net Xpos => lcec.0.0.poscommand
#net Xachse_AF joint.0.amp-enable-out => lcec.0.0.drivecontrol-0
#信号Xachse_AF是任意起名的,如果没定义,net会定义它
net Xachse_AF joint.0.active => lcec.0.0.drivecontrol-0
net Xachse_AF  => lcec.0.0.drivecontrol-3
net Xachse_AF  => lcec.0.0.drivecontrol-5
start
------------------------------------------------------------------

and xml is:
--------------------------------------------
<masters>
	<master idx="0" appTimePeriod="1000000" refClockSyncCycles="5">
	<slave idx="0" type="generic" vid="00100000" pid="000C0108" configPdos="true">

	<sdoConfig idx="6060" subIdx="0">
		<sdoDataRaw data="08"/>
	</sdoConfig>

	
	
	<syncManager idx="0" dir="out"></syncManager>
	<syncManager idx="1" dir="in"></syncManager>

	<syncManager idx="2" dir="out">
		<pdo idx="1600">
			<pdoEntry idx="6040" subIdx="00" bitLen="16" halPin="drivecontrol" halType="bit"/>
			<pdoEntry idx="607A" subIdx="00" bitLen="32" halPin="poscommand" halType="float" scale="1000"/>
		</pdo>
	</syncManager>

	<syncManager idx="3" dir="in">
		<pdo idx="1A00">
			<pdoEntry idx="6041" subIdx="00" bitLen="16" halPin="drivestatus" halType="bit"/>
			<pdoEntry idx="6064" subIdx="00" bitLen="32" halPin="pos" halType="float" scale="1000"/>
		</pdo>
	</syncManager>
	
	<dcConf assignActivate="300" sync0Cycle="*1" sync0Shift="250000"/>
	
	</slave>
	</master>
</masters>
------------------------------------

liuzq71, 2019-05-04:
how to write the hal and xml files to drive a servo in linuxcnc?would you pls modify them to fit it for me?

myself:
hal:
>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
# Erstellt von stepconf am Tue Jul 16 23:06:39 2013
# Änderungen an dieser Datei werden beim nächsten
# Aufruf von stepconf überschrieben.

#loadusr -W lcec_conf /home/pi/linuxcnc/configs/sim.axis/deltaPos.xml
loadusr -W lcec_conf ./is620nPos.xml
loadrt lcec

addf lcec.read-all servo-thread
addf motion-command-handler servo-thread
addf motion-controller servo-thread
addf lcec.write-all servo-thread

#set the control word to 6 and the state will be ready to switch on.
setp lcec.0.0.drivecontrol-1 1
setp lcec.0.0.drivecontrol-2 1

#net Xachse_fb lcec.0.0.pos => joint.0.motor-pos-fb
net Xpos => lcec.0.0.poscommand
#net Xachse_AF joint.0.amp-enable-out => lcec.0.0.drivecontrol-0
#信号Xachse_AF是任意起名的,如果没定义,net会定义它
net Xachse_AF joint.0.active => lcec.0.0.drivecontrol-0
net Xachse_AF  => lcec.0.0.drivecontrol-3
net Xachse_AF  => lcec.0.0.drivecontrol-5
start
>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
and xml is:
>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
<masters>
	<master idx="0" appTimePeriod="1000000" refClockSyncCycles="5">
	<slave idx="0" type="generic" vid="00100000" pid="000C0108" configPdos="true">

	<sdoConfig idx="6060" subIdx="0">
		<sdoDataRaw data="08"/>
	</sdoConfig>

	
	
	<syncManager idx="0" dir="out"></syncManager>
	<syncManager idx="1" dir="in"></syncManager>

	<syncManager idx="2" dir="out">
		<pdo idx="1600">
			<pdoEntry idx="6040" subIdx="00" bitLen="16" halPin="drivecontrol" halType="bit"/>
			<pdoEntry idx="607A" subIdx="00" bitLen="32" halPin="poscommand" halType="float" scale="1000"/>
		</pdo>
	</syncManager>

	<syncManager idx="3" dir="in">
		<pdo idx="1A00">
			<pdoEntry idx="6041" subIdx="00" bitLen="16" halPin="drivestatus" halType="bit"/>
			<pdoEntry idx="6064" subIdx="00" bitLen="32" halPin="pos" halType="float" scale="1000"/>
		</pdo>
	</syncManager>
	
	<dcConf assignActivate="300" sync0Cycle="*1" sync0Shift="250000"/>
	
	</slave>
	</master>
</masters>
>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>

suggestawy, 2019-06-25:
hi,if you get documentation please tell me

sirop, 2019-06-25:
Have a look at https://github.com/koppi/mk/blob/master/Machinekit-Xenomai-Thinkpad-X200.md .
It is in German and uses Xenomai, but it is at least something.