sittner/linuxcnc-ethercat #67: EL6900 configuration

robochime, 2019-04-28:
Good morning Sittner, if you can help me, I'm using EL6900 with two EL1904 inputs and an EL2904 output as your configuration, I guess I have to load the program first through TWINCAT3 in EL6090, but it does not work in linuxcnc, also trying to open the file yours TwinSAFE-project.xml but it does not leave me, can you give me an orientation on this topic ?, thanks.

liuzq71, 2019-05-08:
hi robochime,

i want to write driver for my Inovance IS620N servo imitate lcec_deasda.c, can you give me some suggestions because i find some differences between  IS620N  and asda-a2-e?
thanks

liuzq71, 2019-05-08:
hi sittner,

which series servo is used in swm-fm45a projects? is asda A2-E?thanks

robochime, 2019-05-08:
hello liuzq71, not used any of the two servodriver that you mention, but they are different models and it is normal that it is not the same, I recommend that you write a new modified controller the existing ASDA, you will have to look at the manuals to see the PDO , and how to configure the driver, I can not help you much but I hope it works for you, Regards

closed 2019-05-09 by robochime