sittner/linuxcnc-ethercat #72: Build from source WARNING: ecrt_.........undefined!

nguyenanhtucom, 2019-06-13:
WARNING: "ecrt_master_receive" [/home/nat/linuxcnc-ethercat/src/lcec.ko] undefined!
WARNING: "ecrt_master_create_domain" [/home/nat/linuxcnc-ethercat/src/lcec.ko] undefined!
WARNING: "ecrt_slave_config_sdo" [/home/nat/linuxcnc-ethercat/src/lcec.ko] undefined!
WARNING: "ecrt_master_send" [/home/nat/linuxcnc-ethercat/src/lcec.ko] undefined!
WARNING: "ecrt_domain_queue" [/home/nat/linuxcnc-ethercat/src/lcec.ko] undefined!
WARNING: "ecrt_master_sdo_upload" [/home/nat/linuxcnc-ethercat/src/lcec.ko] undefined!
WARNING: "ecrt_master_sync_slave_clocks" [/home/nat/linuxcnc-ethercat/src/lcec.ko] undefined!
WARNING: "ecrt_slave_config_complete_sdo" [/home/nat/linuxcnc-ethercat/src/lcec.ko] undefined!
WARNING: "ecrt_slave_config_watchdog" [/home/nat/linuxcnc-ethercat/src/lcec.ko] undefined!
WARNING: "ecrt_domain_reg_pdo_entry_list" [/home/nat/linuxcnc-ethercat/src/lcec.ko] undefined!
WARNING: "ecrt_master_application_time" [/home/nat/linuxcnc-ethercat/src/lcec.ko] undefined!
WARNING: "ecrt_master_read_idn" [/home/nat/linuxcnc-ethercat/src/lcec.ko] undefined!
WARNING: "ecrt_domain_process" [/home/nat/linuxcnc-ethercat/src/lcec.ko] undefined!
WARNING: "ecrt_master_callbacks" [/home/nat/linuxcnc-ethercat/src/lcec.ko] undefined!
WARNING: "ecrt_master_state" [/home/nat/linuxcnc-ethercat/src/lcec.ko] undefined!
WARNING: "ecrt_slave_config_state" [/home/nat/linuxcnc-ethercat/src/lcec.ko] undefined!
WARNING: "ecrt_master_slave_config" [/home/nat/linuxcnc-ethercat/src/lcec.ko] undefined!
WARNING: "ecrt_slave_config_dc" [/home/nat/linuxcnc-ethercat/src/lcec.ko] undefined!
WARNING: "ecrt_slave_config_idn" [/home/nat/linuxcnc-ethercat/src/lcec.ko] undefined!
WARNING: "ecrt_domain_data" [/home/nat/linuxcnc-ethercat/src/lcec.ko] undefined!
WARNING: "ecrt_slave_config_pdos" [/home/nat/linuxcnc-ethercat/src/lcec.ko] undefined!
WARNING: "ecrt_master_sync_reference_clock" [/home/nat/linuxcnc-ethercat/src/lcec.ko] undefined!
WARNING: "ecrt_master_deactivate" [/home/nat/linuxcnc-ethercat/src/lcec.ko] undefined!
WARNING: "ecrt_domain_size" [/home/nat/linuxcnc-ethercat/src/lcec.ko] undefined!
WARNING: "ecrt_master_activate" [/home/nat/linuxcnc-ethercat/src/lcec.ko] undefined!
WARNING: "ecrt_release_master" [/home/nat/linuxcnc-ethercat/src/lcec.ko] undefined!
WARNING: "ecrt_request_master" [/home/nat/linuxcnc-ethercat/src/lcec.ko] undefined!
WARNING: "ecrt_slave_config_sdo8" [/home/nat/linuxcnc-ethercat/src/lcec.ko] undefined!

nguyenanhtucom, 2019-06-13:
792.290961] RTAI[math]: loaded.
[  792.437555] lcec: no symbol version for ecrt_slave_config_sdo8
[  792.437559] lcec: Unknown symbol ecrt_slave_config_sdo8 (err -22)
[  792.437562] lcec: no symbol version for ecrt_request_master
[  792.437563] lcec: Unknown symbol ecrt_request_master (err -22)
[  792.437565] lcec: no symbol version for ecrt_release_master
[  792.437567] lcec: Unknown symbol ecrt_release_master (err -22)
[  792.437570] lcec: no symbol version for ecrt_master_activate
[  792.437571] lcec: Unknown symbol ecrt_master_activate (err -22)
[  792.437583] lcec: no symbol version for ecrt_domain_size
[  792.437584] lcec: Unknown symbol ecrt_domain_size (err -22)
[  792.437587] lcec: no symbol version for ecrt_master_deactivate
[  792.437588] lcec: Unknown symbol ecrt_master_deactivate (err -22)
[  792.437593] lcec: no symbol version for ecrt_master_sync_reference_clock
[  792.437594] lcec: Unknown symbol ecrt_master_sync_reference_clock (err -22)
[  792.437596] lcec: no symbol version for ecrt_slave_config_pdos
[  792.437598] lcec: Unknown symbol ecrt_slave_config_pdos (err -22)
[  792.437600] lcec: no symbol version for ecrt_domain_data
[  792.437601] lcec: Unknown symbol ecrt_domain_data (err -22)
[  792.437604] lcec: no symbol version for ecrt_slave_config_idn
[  792.437605] lcec: Unknown symbol ecrt_slave_config_idn (err -22)
[  792.437607] lcec: no symbol version for ecrt_slave_config_dc
[  792.437608] lcec: Unknown symbol ecrt_slave_config_dc (err -22)
[  792.437611] lcec: no symbol version for ecrt_master_slave_config
[  792.437612] lcec: Unknown symbol ecrt_master_slave_config (err -22)
[  792.437618] lcec: no symbol version for ecrt_slave_config_state
[  792.437619] lcec: Unknown symbol ecrt_slave_config_state (err -22)
[  792.437622] lcec: no symbol version for ecrt_master_state
[  792.437623] lcec: Unknown symbol ecrt_master_state (err -22)
[  792.437626] lcec: no symbol version for ecrt_master_callbacks
[  792.437627] lcec: Unknown symbol ecrt_master_callbacks (err -22)
[  792.437629] lcec: no symbol version for ecrt_domain_process
[  792.437631] lcec: Unknown symbol ecrt_domain_process (err -22)
[  792.437634] lcec: no symbol version for ecrt_master_read_idn
[  792.437635] lcec: Unknown symbol ecrt_master_read_idn (err -22)
[  792.437638] lcec: no symbol version for ecrt_master_application_time
[  792.437639] lcec: Unknown symbol ecrt_master_application_time (err -22)
[  792.437641] lcec: no symbol version for ecrt_domain_reg_pdo_entry_list
[  792.437643] lcec: Unknown symbol ecrt_domain_reg_pdo_entry_list (err -22)
[  792.437645] lcec: no symbol version for ecrt_slave_config_watchdog
[  792.437646] lcec: Unknown symbol ecrt_slave_config_watchdog (err -22)
[  792.437650] lcec: no symbol version for ecrt_slave_config_complete_sdo
[  792.437651] lcec: Unknown symbol ecrt_slave_config_complete_sdo (err -22)
[  792.437653] lcec: no symbol version for ecrt_master_sync_slave_clocks
[  792.437655] lcec: Unknown symbol ecrt_master_sync_slave_clocks (err -22)
[  792.437657] lcec: no symbol version for ecrt_master_sdo_upload
[  792.437658] lcec: Unknown symbol ecrt_master_sdo_upload (err -22)
[  792.437662] lcec: no symbol version for ecrt_domain_queue
[  792.437663] lcec: Unknown symbol ecrt_domain_queue (err -22)
[  792.437666] lcec: no symbol version for ecrt_master_send
[  792.437667] lcec: Unknown symbol ecrt_master_send (err -22)
[  792.437669] lcec: no symbol version for ecrt_slave_config_sdo
[  792.437670] lcec: Unknown symbol ecrt_slave_config_sdo (err -22)
[  792.437672] lcec: no symbol version for ecrt_master_create_domain
[  792.437674] lcec: Unknown symbol ecrt_master_create_domain (err -22)
[  792.437677] lcec: no symbol version for ecrt_master_receive
[  792.437678] lcec: Unknown symbol ecrt_master_receive (err -22)
[  799.637191] RTAI[math]: unloaded.
[  799.638144] SCHED releases registered named ALIEN PEDV$D
[  799.643139] RTAI[malloc]: unloaded.
[  799.741293] RTAI[sched]: unloaded (forced hard/soft/hard transitions: traps 0, syscalls 0).
[  799.742461] I-pipe: head domain RTAI unregistered.
[  799.742512] RTAI[hal]: unmounted.
[  803.101203] EtherCAT WARNING 0: 1 datagram TIMED OUT!

sirop, 2019-06-13:
Just switch to rt-preempt for bettter support.

nguyenanhtucom, 2019-06-13:
Thanks sirop 
**Step 1**
create a file "/etc/apt/sources.list.d/linuxcnc-mds.list":
  deb http://repo.modusoft.de/linuxcnc/wheezy ./
# sudo apt-get update
# sudo apt-get install linuxcnc-ethercat

### **Run OK.....**
**Step 2**
Build from source 
git clone https://github.com/sittner/linuxcnc-ethercat.git linuxcnc-ethercat
$ cd linuxcnc-ethercat
$ sudo make  (WARNING: ecrt_.........undefined!)
$ sudo make install
### **Run Error.....(lcec: Unknown symbol ecrt_....)**

### Help me...............!

closed 2019-06-13 by nguyenanhtucom

sirop, 2019-06-13:
Try
````
export CPATH=$CPATH:/usr/local/etherlab/include
echo $CPATH

sudo nano /etc/ld.so.conf.d/ethercat.conf

/usr/local/etherlab/lib # or where esle you installed etherlab

sudo ldconfig
````

then go on with:
````
cd linuxcnc-ethercat
make all
sudo make install
````