sittner/linuxcnc-ethercat #74: igh ethercat driver with beaglebone

lichaowuwei, 2019-07-10:
i am trying to write a w5500driver with machinekit on beaglebone for ethercat, but in kernel 4.9, the slave can't be found. in kerle 3.8 igh could find the slaves using generic driver.
I saw you have build one encx24j600 with machinekit, which kernel do you use?
can you help me to deal with the device tree and driver?
it took me too much time, Thanks Very very Much!

sirop, 2019-07-10:
Do you try to build an EtherCAT slave or an EtherCAT master on your beaglebone?

lichaowuwei, 2019-07-10:
ethercat master on beaglebone
i found that, 4.9 and 4.14 can't find the slaves, both soem and igh.
3.8 kernel is ok

lichaowuwei, 2019-07-10:
> Do you try to build an EtherCAT slave or an EtherCAT master on your beaglebone?

Master, 4.9 and 4.14 kernel can't find slaves, both soem and igh.
but 3.8.13 can ...
don't know why

sirop, 2019-07-10:
Try to build your master with https://github.com/ribalda/ethercat/commits/master which
has patches for kernels 4.x and higher.

lichaowuwei, 2019-07-10:
> Try to build your master with https://github.com/ribalda/ethercat/commits/master which
> has patches for kernels 4.x and higher.

can't work either...
the master code work on kernel 4.x x86 debian, but not beaglebone.

sirop, 2019-07-11:
Guess you read  http://www.machinekit.io/docs/getting-started/machinekit-images-for-bbb/ ?

MouseSun846, 2019-10-06:
Hello everyone! I use linuxcnc to drive ighethercat, my problem, from the slaves hung 6 servo motors, every time the power up to op mode is very slowly, what is the reason?

sirop, 2019-10-06:
Rather try etherlab mailing list and provide more details.

MouseSun846, 2019-10-06:
I don't know if my XML file is set incorrectly, which makes synchronization too slow.

MouseSun846, 2019-10-06:
This is my xml file!
<masters>
		<master idx="0" appTimePeriod="10000000" refClockSyncCycles="2000">
		<slave idx="0" type="generic" vid="00000009" pid="00009252" configPdos="true">
			<sdoConfig idx="1c12" subIdx="complete">
				<sdoDataRaw data="06 00 01 16 11 16 21 16 31 16 41 16 51 16"/>
			</sdoConfig>		
			<sdoConfig idx="1c13" subIdx="complete">
				<sdoDataRaw data="06 00 01 1A 11 1A 21 1A 31 1A 41 1A 51 1A"/>
			</sdoConfig>		
			<sdoConfig idx="6060" subIdx="0">
				<sdoDataRaw data="8"/>
			</sdoConfig>		
			<sdoConfig idx="60C2" subIdx="1">
				<sdoDataRaw data="2"/>
			</sdoConfig>
			<sdoConfig idx="6860" subIdx="0">
				<sdoDataRaw data="8"/>
			</sdoConfig>		
			<sdoConfig idx="68C2" subIdx="1">
				<sdoDataRaw data="2"/>
			</sdoConfig>
			<sdoConfig idx="7060" subIdx="0">
				<sdoDataRaw data="8"/>
			</sdoConfig>		
			<sdoConfig idx="70C2" subIdx="1">
				<sdoDataRaw data="2"/>
			</sdoConfig>
			<sdoConfig idx="7860" subIdx="0">
				<sdoDataRaw data="8"/>
			</sdoConfig>		
			<sdoConfig idx="78C2" subIdx="1">
				<sdoDataRaw data="2"/>
			</sdoConfig>
			<sdoConfig idx="8060" subIdx="0">
				<sdoDataRaw data="8"/>
			</sdoConfig>		
			<sdoConfig idx="80C2" subIdx="1">
				<sdoDataRaw data="2"/>
			</sdoConfig>
			<sdoConfig idx="8860" subIdx="0">
				<sdoDataRaw data="8"/>
			</sdoConfig>		
			<sdoConfig idx="88C2" subIdx="1">
				<sdoDataRaw data="2"/>
			</sdoConfig>
			<syncManager idx="0" dir="out">
      			</syncManager>
			<syncManager idx="1" dir="in">
   			 </syncManager>
			<syncManager idx="2" dir="out">
				<pdo idx="1601">
					<pdoEntry idx="6040" subIdx="00" bitLen="16" halPin="onedrivecontrol" halType="bit"/>
					<pdoEntry idx="607A" subIdx="00" bitLen="32" halPin="oneposcommand" halType="float" scale="1"/>
				</pdo>
				<pdo idx="1611">
					<pdoEntry idx="6840" subIdx="00" bitLen="16" halPin="twodrivecontrol" halType="bit"/>
					<pdoEntry idx="687A" subIdx="00" bitLen="32" halPin="twoposcommand" halType="float" scale="1"/>
				</pdo>
				<pdo idx="1621">
					<pdoEntry idx="7040" subIdx="00" bitLen="16" halPin="threedrivecontrol" halType="bit"/>
					<pdoEntry idx="707A" subIdx="00" bitLen="32" halPin="threeposcommand" halType="float" scale="1"/>
				</pdo>
				<pdo idx="1631">
					<pdoEntry idx="7840" subIdx="00" bitLen="16" halPin="fourdrivecontrol" halType="bit"/>
					<pdoEntry idx="787A" subIdx="00" bitLen="32" halPin="fourposcommand" halType="float" scale="1"/>
				</pdo>
				<pdo idx="1641">
					<pdoEntry idx="8040" subIdx="00" bitLen="16" halPin="fivedrivecontrol" halType="bit"/>
					<pdoEntry idx="807A" subIdx="00" bitLen="32" halPin="fiveposcommand" halType="float" scale="1"/>
				</pdo>
				<pdo idx="1651">
					<pdoEntry idx="8840" subIdx="00" bitLen="16" halPin="sixdrivecontrol" halType="bit"/>
					<pdoEntry idx="887A" subIdx="00" bitLen="32" halPin="sixposcommand" halType="float" scale="1"/>
				</pdo>
			</syncManager>
			<syncManager idx="3" dir="in">
				<pdo idx="1A01">
					<pdoEntry idx="6041" subIdx="00" bitLen="16" halPin="onedrivestatus" halType="bit"/>
					<pdoEntry idx="6064" subIdx="00" bitLen="32" halPin="onepos" halType="float" scale="1"/>
				</pdo>
				<pdo idx="1A11">
					<pdoEntry idx="6841" subIdx="00" bitLen="16" halPin="twodrivestatus" halType="bit"/>
					<pdoEntry idx="6864" subIdx="00" bitLen="32" halPin="twopos" halType="float" scale="1"/>
				</pdo>
				<pdo idx="1A21">
					<pdoEntry idx="7041" subIdx="00" bitLen="16" halPin="threedrivestatus" halType="bit"/>
					<pdoEntry idx="7064" subIdx="00" bitLen="32" halPin="threepos" halType="float" scale="1"/>
				</pdo>
				<pdo idx="1A31">
					<pdoEntry idx="7841" subIdx="00" bitLen="16" halPin="fourdrivestatus" halType="bit"/>
					<pdoEntry idx="7864" subIdx="00" bitLen="32" halPin="fourpos" halType="float" scale="1"/>
				</pdo>
				<pdo idx="1A41">
					<pdoEntry idx="8041" subIdx="00" bitLen="16" halPin="fivedrivestatus" halType="bit"/>
					<pdoEntry idx="8064" subIdx="00" bitLen="32" halPin="fivepos" halType="float" scale="1"/>
				</pdo>
				<pdo idx="1A51">
					<pdoEntry idx="8841" subIdx="00" bitLen="16" halPin="sixdrivestatus" halType="bit"/>
					<pdoEntry idx="8864" subIdx="00" bitLen="32" halPin="sixpos" halType="float" scale="1"/>
				</pdo>
			</syncManager>
			<dcConf assignActivate="300" sync0Cycle="*1" sync0Shift="20000000"/>
		</slave>
		</master>
</masters>

MouseSun846, 2019-10-06:
This is my xml file!

[deltaPos.zip](https://github.com/sittner/linuxcnc-ethercat/files/3693991/deltaPos.zip)

sirop, 2019-10-06:
Hi.

I did not see any obvious errors at once.
Iwould advise you to configer the slave by _ethercat download/upload_ commands instead of raw SDO.

Slowlinessonly is a very vague description of malfunction.
Could youadd some relevant  (kernel) logs?

MouseSun846, 2019-10-06:
Thank you very much for your answer.Command-based information is different from XML configuration。I'm confused.
[image]
[image]
[image]

MouseSun846, 2019-10-06:
I don't know where the problem is.Do I need to slove problem from slaves?

sirop, 2019-10-06:
I do not know either.

1. So begin with ethercat download/upload instead of raw SDO...

MouseSun846, 2019-10-06:
Only one axis can read and write normally, and the rest are read and write failures.

MouseSun846, 2019-10-06:
Another problem is that you can't get into op mode.Error reporting in PreOp mode.

MouseSun846, 2019-10-06:
As shown in the first picture.

curran209, 2021-06-01:
> 我正在尝试在 beaglebone 上为 ethercat 编写带有 machinekit 的 w5500driver，但是在内核 4.9 中，找不到从属设备。在 kerle 3.8 高可以找到使用通用驱动程序的奴隶。
> 我看到你用machinekit构建了一个encx24j600，你用的是哪个内核？
> 你能帮我处理设备树和驱动程序吗？
> 花了我太多时间，非常感谢！

Hello, can I ask you some questions? I can't install the driver now. It was this error after downloading the driver. 
configure.mk:8: *** halcompile/comp executable not found or set.  Stop.
make: *** [config.mk] Error 2
Have you ever encountered it?

curran209, 2021-06-01:
> 如第一张图所示。
我可以请教您一些问题嘛？

lichaowuwei, 2021-06-01:
> Hello, can I ask you some questions? I can't install the driver now. It was this error after downloading the driver.
> configure.mk:8: *** halcompile/comp executable not found or set. Stop.
> make: *** [...]

in the end， i didnot use the driver of w5500, I use the beaglebone's own nic driver.
i modify ti and work pretty good!

good luck to you