sittner/linuxcnc-ethercat #76: patches for linuxcnc - 2.8 / 2.9

robochime, 2019-10-08:
hello sittner could you tell me how to configure the patch for linuxcnc branches 2.8 / 2.9? , Thank you.

narogon, 2019-10-15:
Here are patch files for 2.8 version 2.8.0~pre1. I don't know if they continue working for new versions. 

[patch linux2.8.zip](https://github.com/sittner/linuxcnc-ethercat/files/3729975/patch.linux2.8.zip)

sittner, 2019-10-15:
I think we should try to get the required change to be integrated in the upstream version. I will create a pull request for this on the lcnc git repro this week.

robochime, 2019-10-15:
Hi Naragon, thank you very much for the patch.
Hi Sittner, thank you very much for answering, I know you are very busy, I think it is a good idea, I will try to help in any way I can, thanks.

robochime, 2019-10-31:
Hi, the patches are working fine on the linuxcnc master branch, thanks.

closed 2019-10-31 by robochime