sittner/linuxcnc-ethercat #81: RTAPI_TASK_PLL_SUPPORT not present

narogon, 2019-10-21:
Hi @sittner,

I've just installed new versions. (ec-debianized and hal components). 

But when I load lcec with config file that contains refClockSyncCycles="-1" it says that  RTAPI_TASK_PLL_SUPPORT not present. 
I've just downloaded a fresh linuxcnc 2.7.14 and applied the patch. Configured with` ./configure uspace` , created the .deb packages  and installed like always. 

I'm running #1 SMP PREEMPT RT Debian 4.9.168-1+deb9u4 (2019-07-19)

Any idea of what I'm doing bad? Any way to investigate/solve this? 

I would really appreciate your help because I'm stuck with this and I'm so hopeful this version solve my syncs problems.

narogon, 2019-10-23:
I've achieved run linux with pll but I don't know if something is wrong in my configuration or I miss something. 

In https://github.com/sittner/linuxcnc-ethercat/blob/5399017a1b5887594186aac3722a1a51e0ea07ca/src/lcec_main.c#L395 it checks if RTAPI_TASK_PLL_SUPPORT is defined. 
I've searched where is defined it and discover that it is in rtapi.h after applying the patch. 

`#if defined(RTAPI_USPACE) || defined(USPACE)

#define RTAPI_TASK_PLL_SUPPORT`

But rtapi_uspace or uspace are not defined nowhere but config.h 
I added a #include config.h to lcec_main.c and now pll pin are created and no warning message is printed.

wilsonxiao, 2020-01-04:
Hi, @narogon 
Have you solved this problem？
How to do it? And does it support RTAI Debian?(Linux debian 3.4-9-rtai-686-pae #1 SMP PREEMPT Debian 3.4.55-4linuxcnc i686 GNU/Linux)

narogon, 2020-01-13:
> 
> 
> Hi, @narogon
> Have you solved this problem？
> How to do it? And does it support RTAI Debian?(Linux debian 3.4-9-rtai-686-pae #1 SMP PREEMPT Debian 3.4.55-4linuxcnc i686 GNU/Linux)

I explained in my last paragraph. 

> I added a `#include config.h` to lcec_main.c and now pll pin are created and no warning message is printed.

But I don't know if it is the correct way to solve this.

StormholtVaerks, 2021-04-13:
This worked for me aswell, i dont even know where the config.h is located in my system, i just wrote `#include "config.h"` in lcec_main.c 
My original problem was: undefined symbol: rtapi_task_set_correction

thank you narogon