sittner/linuxcnc-ethercat #87: EL6751 CANOpen configuration in TwinCAT?

marcoreps, 2020-04-27:
Fantastic work on this project!!
I am trying to use an EL6751 CANOpen slice and with ESTUN servo drives that only speak CANOpen. ( https://www.estuneurope.eu/wp-content/uploads/2019/03/Servo-CANOpen-V1.04-Users-Manual_compressed.pdf )

Did I understand that correctly: I have to make a TwinCAT project and configure my EL6751 correctly to generate all the <sdoDataRaw data=".........."/> stuff? ( i.e. in https://github.com/sittner/linuxcnc-ethercat/blob/master/examples/generic-complex/CPD17/ethercat-conf_X.xml )

How can I export the config in TwinCAT then?

sirop, 2020-04-27:
Hi.

Let me put in my 2 cents.

First of all, I doubt that a pure CAN device might be used in any EtherCAT network,
Guess, you will need something like https://esd.eu/en/products/can-ethercat .
There EtherCAT drives that support Can Over EtherCAT (CoE, DS 402) and thus a similar dictionary as 
 CAN-only devices but still CAN over EtherCAT is not the same as CAN.

As for the second question:

TwinCAT config file may bear the same info as the XML file demanded here for a generic slave,
but I doubt that the former and the latter would have the same grammer (structure, tags, etc...).

However the most of the examples in https://github.com/sittner/linuxcnc-ethercat/tree/master/examples are pretty self-explaining. So you can learn by doing.

If the parts with RAW SDO hex code makes you feel any unease, bear in mind
that these RAW SDO pieces can be substituted by  "upload/download" commands of 
EtherLAB Master.

marcoreps, 2020-05-11:
Thank you very much for the EtherLAB Master hint! It will be useful soon.

I have found out where TwinCAT 3 is hiding the EL6751 startup data that I was looking for:
https://imgur.com/3jXXrrX
[image]
( Project Tree -> I/O -> Devices -> Device 2 (EtherCAT) -> Term 1 (EK1100) -> Term 7 (EL6751) and Startup tab on the right )

So this issue, that wasn't really a linuxcnc-ethercat to begin with (sorry about that :) is done. But maybe one more hint if anybody knows: What do I do with that last startup line in my screenshot?

I would guess that it's an InitCmd like https://github.com/sittner/linuxcnc-ethercat/blob/master/examples/initcmds/AM8121-xFx0-000x_MDP.xml
But where do I even start with the syntax? Ccs, Index, Subindex 😅

closed 2020-05-11 by marcoreps