sittner/linuxcnc-ethercat #90: A question about xml_config

ibXCM, 2020-08-31:
When I config the ethercat-conf-X.xml ，there are more than 20 startup SDO in my servo‘s xml。need I config them all？

The list is as following：

0x1C12:00 0x1C13:00 0x1A00:00 0x1A00:01 0x1A00:02 0x1A00:03 0x1A00:04 0x1A00:05 0x1A00:06 0x1A00:07 0x1A00:08 0x1A00:00 0x1A01:00 0x1A01:01 0x1A01:02 0x1A01:03 0x1A01:04 0x1A01:05 0x1A01:06 0x1A01:07 0x1A01:08 0x1A01:09 0x1A01:00 0x1A02:00 0x1A02:01 0x1A02:02 0x1A02:03 0x1A02:04 0x1A02:05 0x1A02:06 0x1A02:07 0x1A02:08 0x1A02:09 0x1A02:00 0x1A03:00 0x1A03:01 0x1A03:02 0x1A03:03 0x1A03:04 0x1A03:05 0x1A03:06 0x1A03:07 0x1A03:08 0x1A03:09 0x1A03:00 0x1600:00 0x1600:01 0x1600:02 0x1600:03 0x1600:04 0x1600:00 0x1601:00 0x1601:01 0x1601:02 0x1601:03 0x1601:04 0x1601:05 0x1601:06 0x1601:07 0x1601:00 0x1602:00 0x1602:01 0x1602:02 0x1602:03 0x1602:04 0x1602:05 0x1602:06 0x1602:00 0x1603:00 0x1603:01 0x1603:02 0x1603:03 0x1603:04 0x1603:05 0x1603:06 0x1603:07 0x1603:08 0x1603:00 0x1C12:01 0x1C12:00 0x1C13:01 0x1C13:00

sirop, 2020-08-31:
Consult the manual of your slave. Normally there would be some configuration examples.

ibXCM, 2020-09-01:
I‘ve download some examples from others。

Thanks