sittner/linuxcnc-ethercat #92: Is it support Linuxcnc 2.9 now？

ibXCM, 2020-09-10:
Dears
       I‘ve tried many times on linuxcnc 2.9 with this driver。but it always has many problems。
       Can it match Linuxcnc 2.9 now？
Thanks & Best Regards
Wade

sirop, 2020-09-10:
Do you run LinuxCNC on linux-rt ?

Could you provide more details about the "problems" ?

ibXCM, 2020-09-11:
Hi Boris
      This is the Error message When I run linuxcnc with my config.

      Starting LinuxCNC...
      Found  file(REL): ./cpd17_x.hal
      Note: Using POSIX realtime
      ./cpd17_x.hal:24: Pin 'lcec.0.2.dout-0' does not exist
      .......

      and My LinuxCNC version is 2.8 . My system is debian9 with kernel
4.19.132-rt
Thanks & Best Regards
Wade

ibXCM, 2020-09-11:
Hi Boris
     The following is my cpd17_x.hal

loadusr -W lcec_conf ethercat-conf_X.xml
loadrt lcec
loadrt trivkins
loadrt [EMCMOT]EMCMOT base_period_nsec=[EMCMOT]BASE_PERIOD
servo_period_nsec=[EMCMOT]SERVO_PERIOD num_joints=[TRAJ]AXES
loadrt not count=1

#-----------------------------------------------------------

addf lcec.read-all servo-thread
addf motion-command-handler servo-thread
addf motion-controller servo-thread
addf not.0 servo-thread
addf lcec.write-all servo-thread

#-----------------------------------------------------------

net bus_up lcec.state-op => iocontrol.0.emc-enable-in

net tool-change  iocontrol.0.tool-change   iocontrol.0.tool-changed
net tool-prepare iocontrol.0.tool-prepare iocontrol.0.tool-prepared
#-----------------------------------------------------------

net Spindel_enable motion.spindle-on => lcec.0.2.dout-0
net Spindel_dir motion.spindle-reverse => lcec.0.2.dout-1

# Test Homefahrt
net RefX halui.spindle.is-on => axis.0.home-sw-in

#-----------------------------------------------------------

# ChangeSetImmediatly: true -> neuer Zielwert sofort aktivieren
setp lcec.0.3.X-cmd-ChangeSetImmediatly true

# NewSetpoint in jedem Zyklus Ã¤ndern
net NewSetpoint not.0.out => not.0.in

#-----------------------------------------------------------
# X-Achse
#-----------------------------------------------------------
net EnableAmpX axis.0.amp-enable-out => lcec.0.3.X-cmd-SwitchOn
net EnableAmpX lcec.0.3.X-cmd-EnableVoltage
net EnableAmpX lcec.0.3.X-cmd-/QuickStop
net EnableAmpX lcec.0.3.X-cmd-EnableOperation
#-----------------------------------------------------------
# Istposition vom Motor an axis Ã¼bergeben
net Xachse_fb lcec.0.3.X-Pos => axis.0.motor-pos-fb
#-----------------------------------------------------------
# Sollposition von axis an den Motor Ã¼bergeben
net Xachse_cmd axis.0.motor-pos-cmd => lcec.0.3.X-TargetPos
#-----------------------------------------------------------
# NewSetpoint an den Motor Ã¼bergeben
net NewSetpoint => lcec.0.3.X-cmd-NewSetpoint

#-----------------------------------------------------------
# Y+Z-Achse simulieren
#-----------------------------------------------------------
net Ypos axis.1.motor-pos-cmd => axis.1.motor-pos-fb
net Zpos axis.2.motor-pos-cmd => axis.2.motor-pos-fb

Thanks & Best Regards
Wade

sirop, 2020-09-11:
>  Starting LinuxCNC... Found file(REL): ./cpd17_x.hal Note: Using POSIX realtime ./cpd17_x.hal:24: Pin 'lcec.0.2.dout-0' does not exist ....... 

This error means that you want to connect a signal a pin called "dout-0" located at the third slave  which in its turn is governed by master  instance 0. So check if all these conditions are true.

ibXCM, 2020-09-11:
Hi，Boris
      I only use one servo to be my Ethercat slave，and I want to make the Y
、Z axis to be virtual。How should I modify my config？
Thanks & Best Regards
Wade

sirop, 2020-09-11:
If you have only 1 master and only 1 slave, then try lcec.0.0.dout-0 and check if _dout-0_ really exists.

To check it, run smth. like:
```
loadusr -W lcec_conf ethercat-conf_X.xml
loadrt lcec
show pin lcec
```

ibXCM, 2020-09-11:
Hi，Boris
      Thanks a million。
      I’ll try it now。
Thanks & Best Regards
Wade

ibXCM, 2020-09-11:
Hi，Boris
       I chenged my .hal like this ,but still have Error messages

*****************************cpd17_x.hal***************************************
# Erstellt von stepconf am Tue Jul 16 23:06:39 2013
# Änderungen an dieser Datei werden beim nächsten
# Aufruf von stepconf überschrieben.
loadusr -W lcec_conf /home/pi/linuxcnc/configs/sim.axis/deltaPos.xml
loadrt lcec
addf lcec.read-all servo-thread
addf lcec.write-all servo-thread

setp lcec.0.0.drivecontrol-1 1
setp lcec.0.0.drivecontrol-2 1

#net Xachse_fb lcec.0.0.pos => axis.0.motor-pos-fb
net Xpos => lcec.0.0.poscommand
net Xachse_AF axis.0.amp-enable-out => lcec.0.0.drivecontrol-0
net Xachse_AF  => lcec.0.0.drivecontrol-3
net Xachse_AF  => lcec.0.0.drivecontrol-5

******************************ERROR********************************************
        HAL: ERROR: thread 'servo-read' not found
        ./cpd17_x.hal:6: addf failed
***********************************************************************************
Thanks & Best Regards
Wade

       and how to run smth? I tried to run smth in my terminal, but it is
bash: smth: command not find

sirop, 2020-09-11:
You have to read more about HAL script language, for instance, how to load a thread.
http://linuxcnc.org/docs/html/hal/tutorial.html#_making_realtime_code_run
smth. means merely something.

ibXCM, 2020-09-11:
Hi，Boris
    Thanks
Thanks & Best  Regards
Wade

ibXCM, 2020-09-14:
Hi, Boris
      I've see the url you send me, and it's very useful.
      I've translate it to our language and share it to my friends.
Thanks & Best Regards
Wade