sittner/linuxcnc-ethercat #93: emc/task/emctaskmain.cc 3087: can't initialize interpreter

ibXCM, 2020-09-15:
Dears
     When I run linuxcnc with my config, there is an error appears as following
LINUXCNC - 2.8.0
Machine configuration directory is '/home/tsinglin/linuxcnc/configs/sim.Test_axis_HC_Sample-2'
Machine configuration file is 'cpd17_x.ini'
Starting LinuxCNC...
Found file(REL): ./cpd17_x.hal
emc/task/emctask.cc 411: interp_error: File not open
File not open
emc/task/emctaskmain.cc 3087: can't initialize interpreter
waiting for s.joints<0>, s.kinematics_type<0>
waiting for s.joints<0>, s.kinematics_type<0>
waiting for s.joints<0>, s.kinematics_type<0>
waiting for s.joints<0>, s.kinematics_type<0>
waiting for s.joints<0>, s.kinematics_type<0>
waiting for s.joints<0>, s.kinematics_type<0>
waiting for s.joints<0>, s.kinematics_type<0>
waiting for s.joints<0>, s.kinematics_type<0>
waiting for s.joints<0>, s.kinematics_type<0>
Shutting down and cleaning up LinuxCNC...

and here is my .hal config，How can I solve it？

loadusr -W lcec_conf ./ethercat-conf_X.xml
loadrt lcec
loadrt trivkins
loadrt [EMCMOT]EMCMOT base_period_nsec=[EMCMOT]BASE_PERIOD servo_period_nsec=[EMCMOT]SERVO_PERIOD num_joints=[TRAJ]AXES
loadrt not count=1

addf lcec.read-all servo-thread
addf motion-command-handler servo-thread
addf motion-controller servo-thread
addf not.0 servo-thread
addf lcec.write-all servo-thread

setp lcec.0.0.ControlWord-1 1
setp lcec.0.0.ControlWord-2 1

#net Xachse_fb lcec.0.0.PositionActualvalue => axis.0.motor-pos-fb
#net Xpos => lcec.0.0.TargetPosition
#net Xachse_AF axis.0.amp-enable-out => lcec.0.0.ControlWord-0
#net Xachse_AF  => lcec.0.0.ControlWord-3
#net Xachse_AF  => lcec.0.0.ControlWord-5

sirop, 2020-09-15:
This error message has nothing to do with linuxcnc-ethercat.
Try https://forum.linuxcnc.org/ instead and close this issue.

ibXCM, 2020-09-15:
Hi，Boris
       Thanks ，I'll go to the linuxcnc forum for help.
Thanks
wade