sittner/linuxcnc-ethercat #94: Unexcepted realtime delay on task 0 with period 1000000

ibXCM, 2020-09-17:
Dears
       When I run linuxcnc,there is always an error exist as following:
-----------------------------------------------------------------
Unexcepted realtime delay on task 0 with period 1000000
This message will only display once per session.
Run the Latency Test and resolve before continuing.
task: main loop took 0.022446 seconds
task: main loop took 0.010904 seconds
task: main loop took 0.011113 seconds
task: main loop took 0.011344 seconds
task: main loop took 0.030612 seconds
Shutting down and cleaning uo LinuxCNC
task: 44863 cycles, min=0.000744, max=0.081036, avg=0.002223, 159 latency excurs
ions (> 10x excepted cycle time of 0.001000s)
Note: Using POSIX realtime
-----------------------------------------------------------------

I'm using debian9 with kernel 4.19.132-rt59. and when I run rt-tests, the max jitter is 90us.
The following is my "demsg" printed

what should I do now？

------------------------------------------------------------------
[16733.892777] ravb e6800000.ethernet eth0: Link is Down
[16733.898024] EtherCAT 0: Link state of ecm0 changed to DOWN.
[16733.903653] EtherCAT WARNING 0: 1 datagram TIMED OUT!
[16733.908741] EtherCAT WARNING 0: 19 datagrams UNMATCHED!
[16733.914019] EtherCAT 0: 0 slave(s) responding on main device.
[16741.137057] ravb e6800000.ethernet eth0: Link is Up - 100Mbps/Full - flow control off
[16741.137876] EtherCAT 0: Link state of ecm0 changed to UP.
[16741.147419] EtherCAT WARNING 0: 1 datagram TIMED OUT!
[16741.362502] EtherCAT 0: 1 slave(s) responding on main device.
[16741.368298] EtherCAT 0: Slave states on main device: INIT.
[16741.374126] EtherCAT 0: Scanning bus.
[16741.577344] EtherCAT 0: Bus scanning completed in 200 ms.
[16741.582790] EtherCAT 0: Using slave 0 as DC reference clock.
[16741.592662] EtherCAT 0: Slave states on main device: PREOP.
[16774.389669] EtherCAT 0: Master thread exited.
[16774.394799] EtherCAT 0: Releasing main device 00:03:82:FE:20:98.
[16774.417715] ec_generic: Unloading.
[16774.444273] EtherCAT: Master module cleaned up.
[16776.520944] EtherCAT: Master driver 1.5.2 unknown
[16776.527372] EtherCAT: 1 master waiting for devices.
[16776.546774] ec_generic: EtherCAT master generic Ethernet device module 1.5.2 unknown
[16776.554641] EtherCAT: Accepting 00:03:82:FE:20:98 as main device for master 0.
[16776.561984] ec_generic: Binding socket to interface 4 (eth0).
[16776.581612] EtherCAT 0: Starting EtherCAT-IDLE thread.
[16776.586977] EtherCAT 0: Link state of ecm0 changed to UP.
[16776.591249] EtherCAT 0: 1 slave(s) responding on main device.
[16776.591259] EtherCAT 0: Slave states on main device: PREOP.
[16776.591672] EtherCAT 0: Scanning bus.
[16776.787515] EtherCAT 0: Bus scanning completed in 196 ms.
[16776.792955] EtherCAT 0: Using slave 0 as DC reference clock.
[16802.516971] EtherCAT: Requesting master 0...
[16802.521311] EtherCAT: Successfully requested master 0.
[16802.528703] EtherCAT 0: Domain0: Logical address 0x00000000, 32 byte, expected working counter 3.
[16802.538527] EtherCAT 0:   Datagram domain0-0-main: Logical offset 0x00000000, 32 byte, type LRW.
[16802.551591] EtherCAT 0: Master thread exited.
[16802.556011] EtherCAT 0: Starting EtherCAT-OP thread.
[16808.125764] EtherCAT WARNING 0-0: Slave did not sync after 5000 ms.
[16808.143958] EtherCAT 0: Domain 0: Working counter changed to 3/3
[16808.143965] .
[16808.272611] EtherCAT 0: Slave states on main device: OP.
[16808.558010] EtherCAT WARNING 0: 14 datagrams UNMATCHED!
[16808.785913] EtherCAT WARNING: Datagram 0000000034347194 (domain0-0-main) was SKIPPED 7 times.
[16809.145873] EtherCAT 0: Domain 0: 4 working counter changes - now 3/3
[16809.145879] .
[16809.559962] EtherCAT WARNING 0: 34 datagrams UNMATCHED!
[16809.789905] EtherCAT WARNING: Datagram 0000000034347194 (domain0-0-main) was SKIPPED 17 times.
[16810.149899] EtherCAT 0: Domain 0: 4 working counter changes - now 3/3
[... 144 lines cut ...]
[16845.279272] EtherCAT 0: Bus scanning completed in 3220 ms.
[16845.285100] EtherCAT 0: Using slave 0 as DC reference clock.
[16845.289855] EtherCAT 0: Domain 0: 4 working counter changes - now 3/3
[16845.289861] .
[16845.559029] EtherCAT WARNING 0: 10 datagrams UNMATCHED!
[16845.933893] EtherCAT WARNING: Datagram 0000000034347194 (domain0-0-main) was SKIPPED 9 times.
[16846.293902] EtherCAT 0: Domain 0: 2 working counter changes - now 3/3
[16846.293909] .
[16846.560366] EtherCAT WARNING 0: 32 datagrams UNMATCHED!
[16846.937895] EtherCAT WARNING: Datagram 0000000034347194 (domain0-0-main) was SKIPPED 16 times.
[16847.297881] EtherCAT 0: Domain 0: 6 working counter changes - now 3/3
[16847.297887] .
[16847.560378] EtherCAT WARNING 0: 34 datagrams UNMATCHED!
[16847.941881] EtherCAT WARNING: Datagram 0000000034347194 (domain0-0-main) was SKIPPED 17 times.
[16848.301898] EtherCAT 0: Domain 0: 6 working counter changes - now 3/3
[16848.301904] .
[16848.560487] EtherCAT WARNING 0: 34 datagrams UNMATCHED!
[16848.945876] EtherCAT WARNING: Datagram 0000000034347194 (domain0-0-main) was SKIPPED 17 times.
[16849.078481] EtherCAT 0: Master thread exited.
[16849.082914] EtherCAT 0: Starting EtherCAT-IDLE thread.
[16849.088856] EtherCAT 0: Releasing master...
[16849.093155] EtherCAT 0: Released.
[16849.097545] EtherCAT WARNING 0-0: Changed to SAFEOP + ERROR in the meantime.
[16849.215460] EtherCAT 0: Slave states on main device: PREOP.
[16849.892503] EtherCAT WARNING: Datagram 000000009709d9b5 (master-fsm) was SKIPPED 1 time.

sirop, 2020-09-17:
First try to undersrtand if this issue is EtherCAT specific or if it comes up with any other LinuxCNC script.

If it is EtherCAT specific, try to understand this https://etherlab-users.etherlab.narkive.com/W9esJT3e/unmatched-and-skipped-datagrams#post2 .

Possible reasons:
1. Mismatch between LinuxCNC thread and EtherCAT cycle times.
2. Latency too high because you use a generic EtherCAT driver.
3. Something else...

ibXCM, 2020-09-17:
Hi，Boris
       Thanks for your reply，I‘ll check it now
Thanks & Best Regards
wade