sittner/linuxcnc-ethercat #96: problem with the integration of the ectty.h library

robochime, 2020-11-08:
Good morning, if you can help me, I integrated the library using #include "../include/ectty.h" the compilation is correct, but when I start lcec.so I have symbols inflicted in ectty_tx_data, I have to modify the realtime.mk file no ? could you help me with that? thanks.

sirop, 2020-11-08:
More info would be helpful.
Please show your installations steps.

robochime, 2020-11-08:
Sirop time,
install ec-debianize with option `--enable-tty`, then modify the tty example to work EL6001, it works fine, now I want to integrate it into linuxcnc-ethercat, created a new one called EL6001, and I can bring it to OP, I want to integrate the library TTY, in the EL6001.C file I include the `"../include/ectty.h"` library and linuxcnc-ethercat compiles well, when I start linuxcnc it gives me the following start lcec.so I have symbols inflicted in ectty_tx_data

sirop, 2020-11-08:
Copy-paste this error message please.

robochime, 2020-11-08:
[image]

sirop, 2020-11-08:
Frankly speaking, I do not know what to do here. Maybe, add _"../include/ectty.h"_ to EXTRA_CFLAGS ...

robochime, 2020-11-08:
Hello @sirop 
not working at the moment, but thank you very much for the help.

robochime, 2020-11-22:
[image]
[image]
[image]

Hello @sirop 
Can I ask you in which file would you add " "../include/ectty.h" a EXTRA_CFLAGS ."

user .mk? realtime.mk?, tried various things seen on the internet, but not which way is correct?

thanks

sirop, 2020-11-23:
You can try user.mk , but what about this:

> I have successfully tested the tty example from ethercat master, but it is
> designed for a kernelspace cyclic task only.
> 
> How can I modify this example to use a Beckhoff EL6002 with my ethercat cyclic
> task in userspace ?

https://etherlab-users.etherlab.narkive.com/4jqS34Oc/virtual-tty-el6002-with-ethercat-in-userspace

robochime, 2020-11-23:
Hello @sirop , if I sent you an email but I have no response, I did not find any example in linuxcnc-ethercat, it may be that you are posing it wrong.
Tanks

sirop, 2020-11-24:
@robochime 
You can try smth. like this, but it is more important to answer first Sebastien Blanchet's question
from https://etherlab-users.etherlab.narkive.com/4jqS34Oc/virtual-tty-el6002-with-ethercat-in-userspace .

robochime, 2020-11-24:
Hello @sirop 
the answer to the question, the virtual TTY only works in KERNEL mode it does not work is USER SPACE

sirop, 2020-11-25:
@robochime 
But you build linuxcnc-ethercat  (lcec.so) for userspace, so as I understand it, you can neither build nor use TTY in userspace.

robochime, 2020-11-25:
Hi @sirop, 
Hi, I'll have to find another way to use serial transmission in user mode, thank you very much

closed 2020-12-02 by robochime