sittner/linuxcnc-ethercat #99: Unable to make

curran209, 2021-05-26:
After downloading the file, you can't make it. What's the problem？

curran209, 2021-05-26:
update-ethercat-config
update-ethercat-config: command not found
What's the problem？

sirop, 2021-05-26:
You mixed up the repos as your issue is about https://github.com/sittner/ec-debianize .

curran209, 2021-05-26:
> 由于您的问题是关于[https://github.com/sittner/ec-debianize的，](https://github.com/sittner/ec-debianize)您混合了回购协议。

First, why can't we "make"?

thank you

sirop, 2021-05-26:
Try to understand https://github.com/sittner/ec-debianize/blob/master/README.debianize#L4-L12 .

And check if  `/usr/sbin` is in your PATH.

curran209, 2021-05-26:
Can it only be executed in Debian system?

> 尝试了解https://github.com/sittner/ec-debianize/blob/master/README.debianize#L4-L12。
> 
> 并检查是否 `/usr/sbin`在您的PATH中。

Can it only be executed in Debian system?