sittner/linuxcnc-ethercat #101: Failed to allocate Memory

satiowadahc, 2021-05-27:
Looking for assistance on this one. 
```bash
halrun
halcmd: loadusr -W lcec_conf ~/conf.xml
Note: Using POSIX realtime                                                                               
Failed to map process data: Cannot allocate memory                                                                
LCEC: failed to activate master 0                                                                             
lcec: rtapi_app_main: Invalid argument (-22)                                                                             
<stdin>:3: waitpid failed /usr/bin/rtapi_app lcec                                                                    
<stdin>:3: /usr/bin/rtapi_app exited without becoming ready                                                          
<stdin>:3: insmod for lcec failed, returned -1
```

```
uname                                                             
5.10.30-rt39 #1 SMP PREEMPT_RT Wed May 19 14:38:10 CST 2021 x86_64 x86_64 x86_64 GNU/Linux
```

I see the error is actually originating from the IGH code, however the driver seems to working on their end.

I can see the master and slaves and communicate with Ethercat commands. But I can't seem to run it in LCNC

satiowadahc, 2021-05-27:
Looks like overflow, master->process_data_size is -120624156508160. but why...

sirop, 2021-05-28:
Could you please post the contents of  your `conf.xml` ...

satiowadahc, 2021-05-28:
```xml
<masters>
  <master idx="0" appTimePeriod="1000000" refClockSyncCycles="-1">
    <slave idx="0" type="EK1101"/>
<!-- <slave idx="0" type="EK1100"/> -->
    <slave idx="1" type="EL1008"/>
  </master>
</masters>
```

I hard coded the master->process_data_size in ethercat lib/master to sizeof(int) and it seems to work now. (until I get more slaves on the network.

satiowadahc, 2021-05-28:
EK1100 works on our older computers running kernel 4.10 for the EK1101, but on my newer laptop I need kernel 5.10 for processor support and tried both the 1101 and 1100. (From pr #102)

sirop, 2021-05-28:
Do you use the source code from the official repo https://gitlab.com/etherlab.org/ethercat  or the unofficial one https://github.com/ribalda/ethercat ?

Do you use generic drivers with kernel 5.10?

Did you try `ethercat debug 1`  for more debug info?

Hard coded slave description works until you get more slaves on the network?

Do you get a core dump of lcec crash?

satiowadahc, 2021-05-28:
1) Official code. 
2) Yes Generic drivers.
3)  Debug is how I found the domain size.

4)
If I set master->process_data_size = 2, I can have 2 slaves or less, I get unmatched datagram errors with more
If I set master->process_data_size = 3, I can have 3 slaves or less, I get unmatched datagram errors with more.
etc.

5) Do  you have a reference or code for getting a core dump?

sirop, 2021-05-31:
You use `refClockSyncCycles="-1"` , which means that you want to sync your master thread to hardware time of a DC capable slave.
On the other hand side you use generic drivers with high latency. This is a bit contradictory.
Correct me if am wrong.

https://github.com/ribalda/ethercat  has some commits for 5.x, although I could find any adopted drivers for kernels newer than 4.19 .