sittner/linuxcnc-ethercat #106: Documentation

satiowadahc, 2021-08-13:


jjrbfi, 2021-08-23:
The documentation looks great.

I'm wondering if you need to change:
      <modParam name="fsoeSlaveIdx" value="3"/>
To this:
      <modParam name="fsoeSlaveIdx" value="2"/>

Because the safety terminals are [ EL1904 idx="2" ] and [ EL2904 idx="4"] .

    
Regards

nbsh2, 2021-09-01:
Thanks! The documentation is useful. But I can't run well on the WAGO 750-354 coupler. It only works when attach it after the EK1100 by the cable. And I didn't know why the WAGO modules can't work independently.

nbsh2, 2021-09-03:
> I find I almost always need an EK1100 in the circuit. Have you checked for out put in dmesg?

No I was't check out the dmesg. But after I remake the source code, the wago module can work independently now. Though I did'n know why it happens.