sittner/linuxcnc-ethercat #111: ASDA-A2-E not responding to vel_cmd (PDO 0x60FF)

Terence-K, 2021-12-02:
I am having a problem getting the servo to move through linuxcnc. I have followed the instructions in the delta ASDA-A2-E manual and can confirm in lcec.deasda.c that everything that should be happening, is happening. I cannot confirm if the data is actually getting written to 0x60FF (target velocity PDO) but due to the enable volts and other control PDO bits working as intended, I have to assume the connection is fine. Seeing others posting problems with tuning problems means they've gotten past the problem I currently have.

Things that work:
-can jog in ASDA_SOFT using usb connection.
- setp lcec.0.x.srv-enable-volts true (enables drive volts in linuxcnc where it can then be followed by lcec.0.x.srv-enable where it then holds torque.
-can read encoder position when enabled or not. Confirmed by spinning motor by hand and seeing the position change in LinuxCNC
-read all the related hal pins to the drive and none of the fault or warnings are true.

Things that don't work
-setp lcec.0.x.srv-vel-cmd 10 ( or any number)

Super frustrating when all the other things work fine including a write to the control PDO, but velocity doesn't even show an error. Its as if the drive isn't looking at that register.

I posted before as tezza120 but the heading might of been too vague?

Any help would be greatly appreciated.

sirop, 2021-12-02:
Why not make up a generic slave XMl file including  0x60FF (target velocity PDO) also as an input pdo entry?

Terence-K, 2021-12-04:
So I've started doing a generic XML as you suggested. Can i add the generic slave as idx ="1" as if it were a second slave, that way keeping all the other functions instead of having to define them all again?

sirop, 2021-12-04:
Do you mean that your slave with idx=0 is already defined as some "hard coded" slave like this https://github.com/sittner/linuxcnc-ethercat/blob/master/examples/generic-complex/ethercat-conf.xml#L3?

idx is real, physikal index , not an alias BTW.

sirop, 2021-12-04:
Besides this you can test the settings of your slave with etherlab sdo commands like `ethercat sdo upload` and `ethercat sdo download`.

Terence-K, 2021-12-04:
Yes, it's hard coded type="DeASDA". I've seen many examples using it and not one with my problem.

xml as follows:

<masters>
<master idx="0" appTimePeriod="1000000" refClockSyncCycles="1000">
	<slave idx = "0" type = "DeASDA" name = "x">
		dcConf assignActivate = "0x300" sync0Cycle = "*1"sync0Shift ="-25000"/>
	</slave>
</master>
</masters>

for some reason lcec_conf does not like having a "<" infront of the dcConf. Removing it lets it run...weird.

So I've been punching in <ethercat upload -p 0 -t int32 0x60FF 0> in terminal while applying torque to the motor, and the number changes in both directions in relation to torque. The feedback is controlling the register and now I know it's being written to.
Jogging in linuxCNC does the same until a following error occurs due to the motor not actually turning. 

I've read the 0x6040, 0x6041, and 0x6060 register and thought I was getting somewhere but the quick stop bit is actually B contact (inverted) so dead end. Something on the servo side is not letting it move. I checked the neg, pos limit switches and Estop - all ok. activating them disables the drive as intended.

Terence-K, 2021-12-05:
Update:

If i change the mode 0x6060 in terminal to profile velocity mode (0x03) and set the accel and decel registers, it will move.
the only realy difference between this mode and CSV is the <code>interpolation / sync0 </code> time relationship which MUST match. I read interpolation time of 1ms on my registers with index of -3. Where can i find the SYNC0 time to compare? XML I've copied shows <code>sync0Cycle = "*1"</code>
pointer to somwhere?

So i have two choices - 1) Go down this path or 2) build a generic XML from scratch to suit Profile velocity  (i'll be guessing my way through this) so i'll check out this SYNC0 thing first and then decide.

Terence-K, 2021-12-05:
Fixed.

So if you have a look at the posted XML above, notice the dcConf without a tag infront of it. Linuxcnc did not like loading it with that tag but when removed, it ingores that line - critical information included.
The problem with that line is no spaces between parameters. The parser did not like that. Needed spaces around the quotes.

So still got following error but it at least moved. changed the scale and dropped the FF1, FF2 values to almost zero and lowered P, moves well by itself. So I'll worry about all those when it's mounted to the machine.

Thanks for your assistance.

sirop, 2021-12-05:
So you found out that lwithin linuxcnc the parser has unexpected behaviour. You applied some tweaks, but still i would not rely
on such a parser.

I have been using this component only within machinekit which means only in user space, and the parser worked as it should.

Mehdidadash, 2022-12-06:
> Fixed.
>
> So if you have a look at the posted XML above, notice the dcConf without a tag infront of it. Linuxcnc did not like loading it with that tag but when removed, it ingores that line - critical [...]

It seems that you solved your problem with asd-a2-E
I cant move motor like you, would you share your xml, hal, ini  here ?
I tried to make servo ready and on yesterday but i was not able to do that using linuxcnc
I used these repositories linuxcnc-ethercat
Hal-cia402 
I think this is because of delta asd-a2-E servo ready procedure
There is this paragraph in its manual
"Set 【Controlword: 6040h】to (0x06 > 0x07 > 0x0F) in sequence, switch the drive to 
Servo On state and enable the motor."
Have you done this ?