sittner/linuxcnc-ethercat #121: supporting separate LRD/LWR commands instead of an LRW command

fdarling, 2023-07-26:
How hard would it be to modify the EtherCAT driver to support forcing separate LRD/LWR commands to be used rather than the combined LRW command? Is this something I could hire someone to work on?

I am trying to use a Yaskawa SGD7S EtherCAT drive with LinuxCNC, and I realized that it is not currently possible due to Yaskawa supporting Logical Read (LRD) and Logical Write (LWR), but _not_ the combo Logical Read Write (LRW) command. This is specified here:

[image]

from this document: [RV-Y4P05Q Beginning EtherCAT Master Development with Yaskawa Drive.pdf](https://www.yaskawa.com/delegate/getAttachment?cmd=docurl&doc=workspace/SpacesStore/fe934263-48fc-43f4-baae-0220c91ffd81/SRV-Y4P05Q%20Beginning%20EtherCAT%20Master%20Development%20with%20Yaskawa%20Drive.pdfRV-Y4P05Q%20Beginning%20EtherCAT%20Master%20Development%20with%20Yaskawa%20Drive.pdf)

This is pretty crazy to me, since the Chinese JMC JASD series servo drives work with LinuxCNC, you'd think Yaskawa would be _more_ compatible!

I posted about this on the LinuxCNC forums:

[Ethercat protocol not responding - Page 3 - LinuxCNC](https://forum.linuxcnc.org/ethercat/48110-ethercat-protocol-not-responding?start=20#275862)

---

You can see how Beckhoff's TwinCAT software has an option "Use LRD/LWR instead of LRW":

[image]

shown here: https://infosys.beckhoff.com/english.php?content=../content/1033/tcsystemmanager/1092577035.html&id=1077609697012163927

and here: https://infosys.beckhoff.com/english.php?content=../content/1033/ethercatsystem/2469132171.html

Or in this case, "Use LRW instead of LWR/LRD":

[image]

shown here: [Schneider Electric Configuration of the EtherCAT Master](https://product-help.schneider-electric.com/Machine%20Expert/V1.1/en/PD.Parameter.LMCEco/PD.Parameter.LMCEco/Configuration_of_the_EtherCAT_Master/Configuration_of_the_EtherCAT_Master.htm)

and here: [Schneider Electric CODESYS Online Help - Tab 'EtherCAT Master - General'](https://product-help.schneider-electric.com/Machine%20Expert/V1.1/en/core_EtherCAT_Configuration_Editor/topics/_ecat_edt_master_master.htm)

fdarling, 2023-08-28:
After a lot of research and experimentation with mostly broken example code, I managed to interface with a Yaskawa EtherCAT servo drive from a standalone C program using [EtherLab.org's "EtherCAT Master" library](https://gitlab.com/etherlab.org/ethercat). This is the same library that the LinuxCNC EtherCAT driver uses, and I was able to apply what I learned to modify that [linuxcnc-ethercat](https://github.com/sittner/linuxcnc-ethercat) driver to have separate read/write domains, and therefore work with Yaskawa EtherCAT.

To get it to work in the short term I disabled compiling the non-generic drivers, as well as disabled some FSoE (Functional Safety over EtherCAT) related code because it was unclear which process data domain it needed to interact with. Once I update the per-device drivers to use the new dual-domain API and fix the FSoE support, I can submit a pull request to get my changes mainlined.

I'll keep you guys posted...

andergisomon, 2025-06-17:
I'm not super lucid on the ETG 1500 description of sub-subdevice comms. Is it really just using any reasonably implemented MainDevice to copy the FSoE MainInstance TxPDO to the FSoE SubInstance RxPDO and vice versa?

hdhedhu04, 2025-12-25:
Hi @fdarling,

Is there any update on this? I am trying to run Yaskawa drives using LinuxCNC EtherCAT. Specifically, I want to know if anyone has been able to create a separate domain in LinuxCNC and run the drives via HAL pins.
If you or anyone else has a working solution, could you please guide me through it? Any pointers, examples, or reference implementations would be very helpful.

Thanks in advance!

fdarling, 2025-12-25:
I have not done anything since I posted last, but as I said, I did get the Yaskawa drives working with LinuxCNC, I just had to remove a bunch of code that really needs to be patched rather than removed. So I can't currently do a pull request since it would break things for other device drivers (which I disabled).

I might try and use OpenCode AI to make the changes, since they are simple but tedious. Basically, every driver that was setting up the PDOs needs to have an extra parameter passed to it since there are separate read/write domains, and it needs to "file" them accordingly. This is not hard, but there are a lot of files that need to be changed! Ideally it would be tested as well, but I cannot do that since I do not have all that hardware...

hdhedhu04, 2025-12-25:
> I have not done anything since I posted last, but as I said, I did get the Yaskawa drives working with LinuxCNC, I just had to remove a bunch of code that really needs to be patched rather than [...]
So for this do you have a repo/git commit of the changes you did for running Yaskawa drives so that I can start the work from there or atleast try to understand.